lgsvl/simulator #376: Can't access Apollo 3.5 docker

vatna, 2019-09-18:
I know that support for 5.0 is now available, but I am doing some work with 3.5 (and 2019.05) and would like to install it on a new PC to continue with that work. When I do `docker pull lgsvl/Apollo-3.5` it downloads a bunch of data but then stops saying `unauthorized: authentication required`. Is there anything I need to do new now to download this?

martins-mozeiko, 2019-09-18:
Our apollo-3.5 docker image is hosted on public docker hub: https://hub.docker.com/r/lgsvl/apollo-3.5
There should be no need to authenticate. Maybe something is wrong with your docker settings?

Currently apollo-3.5 and apollo-5.0 images are exactly the same. So alternatively you could download apollo-5.0 image and rename it to apollo-3.5.

vatna, 2019-09-19:
Thanks! I'll try again, and with the 5.0 image.

closed 2019-09-19 by vatna