lgsvl/simulator #377: WebUI and Build Succesfull but console shows some error

Iqbalparvi, 2019-09-19:
Hi LG team;
I try to build the LGSVL standalone from source using unity ,in unity windows WebUi and  build  successful , but the console show shows some error like AssetBundles not selected like : 
[image]
 I follow the instructions on : 
https://github.com/lgsvl/simulator/blob/master/Docs/docs/build-instructions.md
urgent help required ,
thanks

martins-mozeiko, 2019-09-19:
This error comes from somewhere inside Unity itself and not from our Simulator code. You can safely ignore it, everything will work correctly.

Iqbalparvi, 2019-09-19:
@martins-mozeiko Alright Thanks ,there is another error " No asset bundles can be  selected",
[image]
due to this error i can't see nothing in  the environments and vehicle option before  build, can you have a look  on this screen :
[image]
thanks

martins-mozeiko, 2019-09-19:
That is information message telling that no asset bundles will be built. This is not an error, this is result of a fact that you don't have any environment or vehicle in the project. You need those only when you are modifying map or vehicle. If you are not doing that and are only modifying simulator, you don't need to build bundles, just use existing ones from our [content](https://content.lgsvlsimulator.com) website.

Binary builds of Simulator will download these files automatically, so you don't need to do anything to get maps or vehicles.

Iqbalparvi, 2019-09-20:
@martins-mozeiko thanks now i see the Robotic car in simulator and successfully observe the car movenet in dreamview localhost:8888, but the moving obstacles in the Simulator  can't observe in the dreamview only the car movement  and BeverageAve map shows ,,I trun al modules like prediction,controll,routing,Localization but perception automatically goes off,
any suggestion please

martins-mozeiko, 2019-09-23:
I'm not sure I understand what does not move. Other NPC vehicles?
Could you show screenshot or video recording explaining what does not move?

Iqbalparvi, 2019-09-25:
@martins-mozeiko thanks for support,
basicllay i run all the steps well from initial to bootstrap.sh and bridge.sh in terminal with apollo -5.0 as:
[image]
then i run localhost:8888 in the browser and see the dreamview and open all the basic modules as : 
[image]
at the same time i run the LGSVL Simulator and See the Robot car and all the other vehicles in the surroundings as:
[image]
when the robot car move in the simulator then the car in the dreamview also move 
but i can't see the surrounding vehicles in the dreamview  and when I set the routing points  it shows the red line thats good but the car can't stop in the ending point and moving still it crash at somewhere:
[image]
so ,what is the problem here ,why i can't see the obstacles in the dreamview and why the car can't stop on that point that we set by editing in the dreamview?
Regards

martins-mozeiko, 2019-09-25:
You need to enable Perception, Traffic Light and Tranform modules for Apollo to see other vehicles.

If these modules do not turn on, then check logfile - under apollo data/log folder. Typically these modules do not work because they cannot initialize GPU.

Check that `nvidia-smi` reports your GPU correctly when run inside docker.

Are you using Apollo supported GPU? They officially support only GTX 1080. But similar generation GPU's will also work (1060, 1070). Newer GPUs will **not** work - Titan X, RTX 2080 and similar.

Iqbalparvi, 2019-09-25:
yes there is no files or information present inside data/log file in apollo ,so how can i slove this issue ?
yes I used GPU GTX 1060 that is still enough to support apollo.
 to check the nvidia-smi outside docker it shows : 
[image]
but when i go inside the dev and run the command to intsall nvidia as:
1. wget http://us.download.nvidia.com/XFree86/Linux-x86_64/418.56/NVIDIA-Linux-x86_64-418.56.run
 2. sudo bash ./NVIDIA-Linux-x86_64-418.56.run --no-x-check -a -s --no-kernel-module
it shows error at the final stages like:
[image]
so is it possible that  this nvidia-smi create the obstacles observation problem in the dreamview?
kindly suggest any  proper solution to solve the above issue
thanks

martins-mozeiko, 2019-09-25:
Ah, I forgot that Apollo does not have nvidia-smi.
Try installing mesa-demos: `sudo apt install mesa-utils`
And then run `glxinfo`
It will show many lines which should have something like this in the middle:

```
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1080/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 435.21
```

If it does not show these, then your nvidia-docker is not correctly installed and perception modules do not have access to GPU.

As for log files - try deleting whole `/apollo/data` folder, maybe it is permission issue and Apollo cannot write. It will automatically recreate this folder.

Iqbalparvi, 2019-09-26:
@martins-mozeiko ,i run the command in terminal as sudo apt install mesa-utils which success: 
[image]
and then run the command glxinfo that shows (think the Nvdia docker is installed):
[image]
at the same time i delete the apollo/data/log folder and restart the machine ,but the problem also reamains same ,the perceptio and traffic light automatically enable and no obstacles can be viewed in the dreamview ,so I dontknow what is the problem behind this ,why the perception modules cant work?

martins-mozeiko, 2019-09-26:
You need to get log files for these modules when they fail. Files should be written in "data/log" folder. Otherwise it is hard to understand what is failing here.
If you cannot get Apollo running, I can suggest asking support in Apollo github issues: https://github.com/ApolloAuto/apollo/issues They probably can provide more detailed troubleshooting.

Iqbalparvi, 2019-10-02:
@martins-mozeiko thanks for your suggestions ,unfortunately i loss all my previous data so i start build lgsvl again but my build fail and console shows the error: 
[image]
can you please suggest me any solution to handle this issue
regards

martins-mozeiko, 2019-10-02:
These are harmless errors/warnings from internal components of Unity. You can safely ignore these. If there is no more error messages, then build should be successful.

Iqbalparvi, 2019-10-02:
but here i see the message Build failed  due to error building players 4 errors ,and there is no simulator inside the build folder.....

martins-mozeiko, 2019-10-02:
Do you have vulkan client libraries installed? Check what title Unity Editor has. It should have `Vulkan` in it. If it does not have that, and instead it has `OpenGL` then you need to install vulkan client libraries for your GPU.

Iqbalparvi, 2019-10-02:
the Unity editor looks like as in the unity -2019.1.10f1 folder 
[image]
but i dont know how to check Vulkan here  and how to install it ..?

martins-mozeiko, 2019-10-02:
You should check title of Unity Editor window for `OpenGL` or `Vulkan` string:
[image]

Installing vulkan client libraries is different depending on GPU and drivers you use. If you use propriatary nvidia binary driver, then you need to run `sudo apt install libvulkan1`

Iqbalparvi, 2019-10-02:
@martins-mozeiko thanks mate you save my day thanks alot ...

Iqbalparvi, 2019-10-02:
@martins-mozeiko  after a big fight as Build success but when i click on the Simulator it  bring me the localhost:8080 but there is nothing to run although my WebUI build also succes without any error:
[image] 
is there any other posible mistakes that cause this issue?
thanks

closed 2019-10-03 by Iqbalparvi