lgsvl/simulator #379: 2019.09, when I build the simulator in Unity, there are no environments and vehicles? and when i open the browser there are no maps vehicles and simulations?

gaohongfein, 2019-09-19:


gaohongfein, 2019-09-19:
[image]
[image]

srirambadri, 2019-09-19:
Same error. Missing assets

srirambadri, 2019-09-19:
@gaohongfein try downloading it from here https://content.lgsvlsimulator.com/

srirambadri, 2019-09-19:
But unity is not able to detect them
[image]

gaohongfein, 2019-09-19:
> @gaohongfein try downloading it from here https://content.lgsvlsimulator.com/

Have you solve this problem? I can not run the Unity without root. This can result in this problem?

srirambadri, 2019-09-19:
Even without root unity still doesn't recognize the files. I'm having the same problem as you

srirambadri, 2019-09-19:
Look at this issue 
https://github.com/lgsvl/simulator/issues/318?_pjax=%23js-repo-pjax-container

gaohongfein, 2019-09-19:
I can't also recognize these files。。。

gaohongfein, 2019-09-19:
I can't open the browser after I click button“open browser"
[image]

martins-mozeiko, 2019-09-19:
The maps and vehicles under Build menu needs to be added manually to project. You need to build them only if you are modifying environment or vehicle. If you are modifying only simulator you can use binary map and vehicle bundles that you can get on our content website. Binary build of simulator will automatically download these maps and vehicles.

Read more information on asset building in our documentation: https://www.lgsvlsimulator.com/docs/assets/

srirambadri, 2019-09-20:
@martins-mozeiko hi thanks for the info. I got the binaries from the https://content.lgsvlsimulator.com/, and put them in the folders as specified in the link you had suggested. But I still get an error saying the ".scenefile" are missing. Do you know where to download the correct binaries from?
[image]

martins-mozeiko, 2019-09-20:
I think you misunderstood me.
You need to put git cloned folders in `External/{Environments,Vehicles}` folder. For example, SingleLaneRoad repository from https://github.com/lgsvl/SingleLaneRoad/

You do that only when you are modifying map or vehicle. If you are not then, there is no need to build anything. Asset bundles from content website are already built by us. These are final binaries. You need to add them in web interface directly:
https://www.lgsvlsimulator.com/docs/maps-tab/#how-to-add-a-map
https://www.lgsvlsimulator.com/docs/vehicles-tab/#how-to-add-a-vehicle

srirambadri, 2019-09-23:
That's exactly what I was looking for, thanks.

closed 2019-10-02 by martins-mozeiko