lgsvl/simulator #380: some error when i open the simulator

gaohongfein, 2019-09-19:
Preloaded type GtkListStore
Preloaded type GtkWindow
Preloaded type GtkVBox
Preloaded type GtkImage
Preloaded type GtkNotebook
Preloaded type GtkHBox
Preloaded type GtkFrame
Preloaded type GtkAlignment
Preloaded type GtkTreeView
Preloaded type GtkLabel
Preloaded type GtkCheckButton
Preloaded type GtkScrolledWindow
Preloaded type GtkComboBox
Desktop is 1920 x 1080 @ 60 Hz
[Vulkan init] extensions: count=14
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_KHX_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
Vulkan detection: 2
Initialize engine version: 2019.1.10f1 (f007ed779b7a)
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=14
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_KHX_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: count=32
[Vulkan init] extensions: name=VK_KHR_swapchain, enabled=1
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_memory_requirements2, enabled=1
[Vulkan init] extensions: name=VK_KHR_maintenance1, enabled=1
[Vulkan init] extensions: name=VK_KHR_push_descriptor, enabled=0
[Vulkan init] extensions: name=VK_KHR_sampler_mirror_clamp_to_edge, enabled=1
[Vulkan init] extensions: name=VK_KHR_shader_draw_parameters, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_fd, enabled=0
[Vulkan init] extensions: name=VK_KHX_device_group, enabled=0
[Vulkan init] extensions: name=VK_KHX_external_memory, enabled=0
[Vulkan init] extensions: name=VK_KHX_external_memory_fd, enabled=0
[Vulkan init] extensions: name=VK_KHX_external_semaphore, enabled=0
[Vulkan init] extensions: name=VK_KHX_external_semaphore_fd, enabled=0
[Vulkan init] extensions: name=VK_KHX_multiview, enabled=0
[Vulkan init] extensions: name=VK_EXT_discard_rectangles, enabled=0
[Vulkan init] extensions: name=VK_EXT_shader_subgroup_ballot, enabled=0
[... 176 lines cut ...]
UnityEngine.EventSystems.EventSystem:Update()
 
(Filename: <5a2a7e03f4364fc6a8d787414ad33e8c> Line: 0)

NullReferenceException: Object reference not set to an instance of an object
  at LoaderUI.OnStartButtonClick () [0x00005] in <5a2a7e03f4364fc6a8d787414ad33e8c>:0 
  at UnityEngine.Events.InvokableCall.Invoke () [0x00011] in <788595b234544a53a21c9f0449515c8f>:0 
  at UnityEngine.Events.UnityEvent.Invoke () [0x00023] in <788595b234544a53a21c9f0449515c8f>:0 
  at UnityEngine.UI.Button.Press () [0x00027] in <df152ef7237747c6aeec0d5524108ec4>:0 
  at UnityEngine.UI.Button.OnPointerClick (UnityEngine.EventSystems.PointerEventData eventData) [0x00011] in <df152ef7237747c6aeec0d5524108ec4>:0 
  at UnityEngine.EventSystems.ExecuteEvents.Execute (UnityEngine.EventSystems.IPointerClickHandler handler, UnityEngine.EventSystems.BaseEventData eventData) [0x00008] in <df152ef7237747c6aeec0d5524108ec4>:0 
  at UnityEngine.EventSystems.ExecuteEvents.Execute[T] (UnityEngine.GameObject target, UnityEngine.EventSystems.BaseEventData eventData, UnityEngine.EventSystems.ExecuteEvents+EventFunction`1[T1] functor) [0x00070] in <df152ef7237747c6aeec0d5524108ec4>:0 
UnityEngine.DebugLogHandler:Internal_LogException(Exception, Object)
UnityEngine.DebugLogHandler:LogException(Exception, Object)
UnityEngine.Logger:LogException(Exception, Object)
UnityEngine.Debug:LogException(Exception)
UnityEngine.EventSystems.ExecuteEvents:Execute(GameObject, BaseEventData, EventFunction`1)
UnityEngine.EventSystems.StandaloneInputModule:ReleaseMouse(PointerEventData, GameObject)
UnityEngine.EventSystems.StandaloneInputModule:ProcessMousePress(MouseButtonEventData)
UnityEngine.EventSystems.StandaloneInputModule:ProcessMouseEvent(Int32)
UnityEngine.EventSystems.StandaloneInputModule:ProcessMouseEvent()
UnityEngine.EventSystems.StandaloneInputModule:Process()
UnityEngine.EventSystems.EventSystem:Update()
 
(Filename: <5a2a7e03f4364fc6a8d787414ad33e8c> Line: 0)

martins-mozeiko, 2019-09-19:
```SqliteException: Attempt to write a read-only database
attempt to write a readonly database
```

This looks like an error with access to database Simulator uses.
Have you run Simulator as root? Maybe it created file with wrong permissions and now your regular user does not have access to it?

Check database file owner and permissions. It is in `~/.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator/data.db` location.

If nothing helps, just delete whole `~/.config/unity3d/LG Silicon Valley Lab` folder. It will be automatically recreated next time you run Simulator.

gaohongfein, 2019-09-20:
[image]
[image]
I have open the browser successfully, but there are nothing environment, no map vehicles.

martins-mozeiko, 2019-09-20:
Is this first time you launched simulator? Or have you opened it also in Unity Editor?
If you will open project in Unity Editor then it will not download default maps/vehicles automatically.
It will download default assets only when binary build runs first.

Alternative is to simply add map/vehicle assets manually from our content website: https://content.lgsvlsimulator.com

gaohongfein, 2019-09-20:
Yes the first time, what should I download, it doesn't run in Unity Editor.

gaohongfein, 2019-09-20:
After I download, in which folder the Environment and 
Vehicles should be put?

gaohongfein, 2019-09-20:
[8251:8265:0920/131958.196320:ERROR:ssl_client_socket_impl.cc(943)] handshake failed; returned -1, SSL error code 1, net_error -101
[8251:8265:0920/131958.196384:ERROR:ssl_client_socket_impl.cc(943)] handshake failed; returned -1, SSL error code 1, net_error -101
 what is this mean?

EricBoiseLGSVL, 2020-04-07:
@gaohongfein closing this issue, please open another if you have any further problems

closed 2020-04-07 by EricBoiseLGSVL