lgsvl/simulator #381: I can't open the browser after I click button“open browser"

enginBozkurt, 2019-09-19:
I can't open the browser after I click button“open browser"
[image]

_Originally posted by @gaohongfein in https://github.com/lgsvl/simulator/issues/379#issuecomment-533139685_

enginBozkurt, 2019-09-19:
Assets and Web UI are built. I tired localhost:8080  but keeps reloding the same page constantly.

martins-mozeiko, 2019-09-20:
What URL is show in web interface? Is it http://localhost:8080/ directly, or is ending with /#/Maps or similar? We have known issue that only opening http://localhost:8080/ URL directly works, you cannot refresh other URLs.

Roman-jj, 2019-09-20:
#360 There's a similar issue here. Are you trying to run the WebUI behind a proxy?

enginBozkurt, 2019-09-20:
> #360 There's a similar issue here. Are you trying to run the WebUI behind a proxy?

I am not using any proxy, especially I had read this issue and canceled the proxy.

enginBozkurt, 2019-09-20:
> What URL is show in web interface? Is it http://localhost:8080/ directly, or is ending with /#/Maps or similar? We have known issue that only opening http://localhost:8080/ URL directly works, you cannot refresh other URLs.

[image]

It is only http://localhost:8080/ . After entering creditentials (email and password), web browser automatically refreshes itself and not connecting the Web Ui.

xmfcx, 2019-10-16:
To reproduce the bug, add 
`Thread.CurrentThread.CurrentCulture = new CultureInfo("tr-TR"); `
to https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Web/Loader.cs#L115
Or use a windows pc with Turkish Locale settings.

And when you run it, it will say 
`SqliteException: SQLite error

no such column: maps.ıd
`
and it won't open a browser window with the button.
It is deserializing Id to ıd at some point. Iı İi

To fix it, add 
`Thread.CurrentThread.CurrentCulture = new CultureInfo("en-US"); `
to https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Web/Loader.cs#L115
and necessary using statements to the top of the file. 
`
using System.Globalization;
using System.Threading;
`

martins-mozeiko, 2019-10-16:
Thanks, I can reproduce this issue. We'll look into fixing it.

martins-mozeiko, 2019-10-17:
Here is the change that will fix this:
```
 Assets/Scripts/Database/DatabaseManager.cs | 9 +++++++--
 1 file changed, 7 insertions(+), 2 deletions(-)

diff --git a/Assets/Scripts/Database/DatabaseManager.cs b/Assets/Scripts/Database/DatabaseManager.cs
index 1f2c419a..6c5ae2c6 100644
--- a/Assets/Scripts/Database/DatabaseManager.cs
+++ b/Assets/Scripts/Database/DatabaseManager.cs
@@ -43,6 +43,11 @@ namespace Simulator.Database
             return $"Data Source = {GetDatabasePath()};version=3;";
         }
 
+        static string UncapitaliseInvariant(string word)
+        {
+            return word.Substring(0, 1).ToLowerInvariant() + word.Substring(1);
+        }
+
         public static IDatabaseBuildConfiguration GetConfig(string connectionString)
         {
             return DatabaseConfiguration.Build()
@@ -51,10 +56,10 @@ namespace Simulator.Database
                 .UsingDefaultMapper<ConventionMapper>(m =>
                 {
                     // Vehicle => vehicles
-                    m.InflectTableName = (inflector, tn) => inflector.Pluralise(inflector.Uncapitalise(tn));
+                    m.InflectTableName = (inflector, tn) => inflector.Pluralise(UncapitaliseInvariant(tn));
 
                     // TimeOfDay => timeOfDay
-                    m.InflectColumnName = (inflector, cn) => inflector.Uncapitalise(cn);
+                    m.InflectColumnName = (inflector, cn) => UncapitaliseInvariant(cn);
                 });
         }
 ```

We'll include it in next push to github.

exoticDFT, 2020-03-14:
I just discovered this simulator and am running into the same issue after creating an account and logging in. I'm using the 2020.01 release binaries on Fedora 31.

EricBoiseLGSVL, 2020-04-07:
@enginBozkurt @exoticDFT We will look into this issue

closed 2020-10-06 by EricBoiseLGSVL