lgsvl/simulator #383: windows standalone build

david-gwa, 2019-09-20:
hey, lg-team,

     we have a little problem with lg-sim windows build.

 option 1), build the project  in Unity Editor env directly, it won't generate the AssetBundle folder

option 2), build with the [command line](https://github.com/lgsvl/simulator/blob/master-obsolete/Docs/docs/build-instructions.md), it generates the AssetBundle folder, but it's empty inside.

we had Ubuntu command line built, works well.

martins-mozeiko, 2019-09-20:
Hi.
Does the Build window show any environments or vehicles and you have selected them?
If it does not show any environments there, then please make sure you have put them in correct locations. Read about asset building in our documentation: https://www.lgsvlsimulator.com/docs/assets/

david-gwa, 2019-09-20:
sorry, need to clarify, we are still in the old version, and prefer the command line build, and during run the build commands, there is no Build windows showup.  

after built, the folder looks like: 

[image]

`simulator.exe` binary is working, but `AssetBundles` folder is empty, which gives null list of map:

[image]

martins-mozeiko, 2019-09-20:
For old simulator version see comments in following issue: https://github.com/lgsvl/simulator/issues/41

david-gwa, 2019-09-20:
checked issue: #41,  and my built actually also work well in Editor Run:

[image]

while the standalone binary in non Editor env, doesn't have AssetBundle.  looks like AssetBundleManager.prefab is not triggered when `build in command line`

EricBoiseLGSVL, 2020-04-07:
@david-gwa yes, Editor loads from scene file while build loads from asset bundle.  Please update to latest simulator for latest features and fixes

closed 2020-04-07 by EricBoiseLGSVL