lgsvl/simulator #384: Npc vehicles do not stop at traffic intersections

liuppppp, 2019-09-23:
hi, lg team
I created a simple traffic light in the SimpleMap scene. I made some manual settings with reference to existing traffic intersections, including signal segment and stop line. like this:
[image]

There is a problem
The npc vehicle does not respond at all to this traffic light. When it is red, npc will not stop.

What is wrong with me? thx

EricBoiseLGSVL, 2019-09-24:
@liuppppp 

Let's see if I can help.

1) Make sure Intersection object has IntersectionComponent.cs attached.  This has a yield trigger radius public value. Make sure when in play mode it is inside the stoplines for the intersection.

2) Be sure that TrafficLightLL object has IntersectionTrafficLightSetComponent.cs attached and the Light renderers list is correct (the child meshes) and the StopLine is correctly set.

3) Also, be sure that the HDMapSignalLight has the MapSignalLightBuilder.cs attached and HintStopline correctly assigned.

When you duplicate objects in Unity and there are scripts attached, you may break the public serialized variables.  You need to manually drag those in the inspector.  An easy way to check if they are correct is too left click the field and unity will highlight the object in the Hierarchy Panel.  Check how the other intersections are connected by trying the left click method.  The scripts need to reference the child objects of that intersection, not the duplicated from child objects.

Let me know if you have more questions or issues.

EricBoiseLGSVL, 2020-04-07:
@liiiu closing issue. if you have any other problems please open a new issue

closed 2020-04-07 by EricBoiseLGSVL