lgsvl/simulator #385: How to pub sensor topic in ROS?

Oasis-Guo, 2019-09-23:
Dear developers,

I am using the newest version of LGSVL. I try to create the ego vehicle from source. But when I build the assert bundle and try to use it from simulator, the sensor topic(lidar, imu, etc) didn't publish to ros, the ros bridge is connected and sensor is enable.

I know that the assest/script in newest version seems change, compare to 2019.05 version.
At 2019.05 version, using AgentSetup script could do that, but now the script is missing at the newest version.

hadiTab, 2019-09-23:
Have you set up the sensor configuration for the vehicle in the web UI?
We have [documentation](https://www.lgsvlsimulator.com/docs/sensor-json-options/) with examples for different sensors (including lidar and imu).

Oasis-Guo, 2019-09-24:
> Have you set up the sensor configuration for the vehicle in the web UI?
> We have [documentation](https://www.lgsvlsimulator.com/docs/sensor-json-options/) with examples for different sensors (including lidar and imu).

Thanks!!!!!!! It solve my problem

closed 2019-10-02 by martins-mozeiko