lgsvl/simulator #386: How can you create an accurate map?

ninjacoder0, 2019-09-23:
I would like to create an accurate map using software like Here Map or any other accurate mapping software you suggest which would give me precise mapping. I would like to place it into Unity where I can then put on the lanes and run it in Apollo.

What would you suggest. 
Thanks all.

david-gwa, 2019-09-25:
for q1, lg-sim is starting to support(confirm me) importing  OpenDrive(lane2) similar map format, so if your Here map can transfer  into these format, should be achieveable.

ninjacoder0, 2019-09-25:
Great, is there any documentation or guidance of how this is done?

martins-mozeiko, 2019-09-25:
If all you need is to get HD map information in our Simulator, you can use `Simulator -> Import HD Map` menu item. Currently it supports importing Lanelet2 format. We are working un adding Apolo and OpenDRIVE imports.

Note that this will add only lane, traffic light, intersection annotations. Creating 3D geometry is additional work you'll need to do. In simplest case you can add flat plane inside Unity Editor where car can drive. If you need more sophisticated geometry - curbs, grass, buildings, then all that needs to be created with 3rd party tools.

As for communication with Apollo - we send exactly same messages as real Apollo vehicle does. This includes lidar, multiple cameras, gps, imu, canbus messages. You can find definition of these messages in protobuf files inside Apollo repostiory. Or just look them up in Simulator C# code. You can also use in-simulation GUI to see what topics are being published: https://www.lgsvlsimulator.com/docs/bridge-connection-ui/

closed 2020-04-07 by EricBoiseLGSVL