lgsvl/simulator #388: Can not build WebUI

wangshuolot, 2019-09-23:
i have finished step22 at https://www.lgsvlsimulator.com/docs/build-instructions/.
when i run Build WebUi... in Unity: Simulator -> Build WebUI...,
There are some error in my unity editor
 
1:Win32Exception: The system cannot find the file specified.

System.Diagnostics.Process.StartWithShellExecuteEx (System.Diagnostics.ProcessStartInfo startInfo) (at <0079a30f96a047348857e1cecc6c638a>:0)
System.Diagnostics.Process.Start () (at <0079a30f96a047348857e1cecc6c638a>:0)
(wrapper remoting-invoke-with-check) System.Diagnostics.Process.Start()
Simulator.Editor.BuildWebUI.Build () (at Assets/Scripts/Editor/BuildWebUI.cs:32)
UnityEngine.Debug:LogException(Exception)
Simulator.Editor.BuildWebUI:Build() (at Assets/Scripts/Editor/BuildWebUI.cs:39)

2:Failed to build WebUI. Please make sure npm is installed. Check README.md file

UnityEngine.Debug:LogError(Object)
Simulator.Editor.BuildWebUI:Build() (at Assets/Scripts/Editor/BuildWebUI.cs:40)

and i run simulator check
there is an error:ERROR: File '/Assets/Scripts/Editor/OpenDRIVE_1.4H.cs' does not have correct copyright header

martins-mozeiko, 2019-09-23:
Are you sure you have npm installed? Can you try run "npm" in commandline, does it succeed?

wangshuolot, 2019-09-24:
> Are you sure you have npm installed (from step 7)? Can you try run "npm" in commandline, does it succeed?

npm have installed,and i run "npm install" already. i have gotten a new folder "node_modules"
then i run Build WebUi and i got a error sadly

martins-mozeiko, 2019-09-24:
You can try running `npm run pack` inside WebUI folder in commandline. This is what Build WebUI menu item is running. It will probably show more detailed error message.

wangshuolot, 2019-09-25:
> You can try running `npm run pack` inside WebUI folder in commandline. This is what Build WebUI menu item is running. It will probably show more detailed error message.

Thanks for your help, i can build my project!

closed 2019-09-25 by wangshuolot

Bnew1823, 2019-09-27:
I'm faced with the exact same problem.
May I ask you how you solved it?
I ran the npm run pack.

> simulator-web-app@1.0.0 pack C:\Users\CrossDesignLab\simulator\WebUI
> enact pack

Creating a development build...
        4.93 MB         dist\main.js
        34.17 KB        dist\main.css

Compiled with warnings:

./src/components/SimulationManager/FormMapVehicles.js
  Line 126:  'vehicleList' is already declared in the upper scope  no-shadow

NOTICE: This build contains debugging functionality and may run slower than in production mode.

Bnew1823, 2019-09-27:
[image]

martins-mozeiko, 2019-09-27:
@Bnew1823 
This is output of successful build. If you see this the everything build correctly.
What exactly does not work after build finished?

Bnew1823, 2019-09-30:
@martins-mozeiko 
It works when I ran the npm run pack
However, this error still occurs when I run build WebUI in unity.
[image]
'지정된 파일을 찾을 수 없습니다' means 'no specified file found'

And nothing appears in the building maps& vehicles window.

martins-mozeiko, 2019-09-30:
First two are not real errors. This is something Unity likes to show when it is in progress of opening/importing new project.

Last one is error about npm not found. Please check that it is correctly set in path. If you are not planning to change WebUI, then you can simply ignore building it from Unity Editor. Building it just once from command-line is enough.

Bnew1823, 2019-09-30:
When I start, it run through the web, but there are some errors unlike when it act as a simulator.
For example, it is always night, even though time is controlled by day.
And I want to edit Scene in Unity.
I'm looking for a scene with a map before I run it, but I don't know where it is.

Bnew1823, 2019-09-30:
And I have a question. Can I get a map of San Francisco from Unity without using Apollo?

Julyers, 2020-03-24:
> @martins-mozeiko
> It works when I ran the npm run pack
> However, this error still occurs when I run build WebUI in unity.
> [image]
> '지정된 파일을 찾을 수 없습니다' means 'no specified file found'
> 
> And nothing appears in the building maps& vehicles window.

So how did you solve it in the end? I also encountered the same problem.