lgsvl/simulator #390: Rosbridge Sensor delay issue

Oasis-Guo, 2019-09-25:
Hi, 

currently I am using rosbridge to connect ROS and LGSVL and get the LIDAR message to visualize it at Rviz. Here is what I found:

1) If I use 10Hz Lidar, then after running for like 5 minutes, the Lidar topic would have some delay, jump or flash,  and it is very hard to recover back, since I can see it from Rviz.  

2)if I use 5-8Hz Lidar without camera, it looks fine for a lone time running.

I check online, it is said that this could be happen from rosbridge queue up its message when message flow is too huge. I try to relaunch the rosbridge when this happen, and it does recovery back. I just wondering is there any way to improve this? And is this could be solve by upgrade my hardware?

My running envirnoment info is:
1.2019.09 LGSVL
2.ROS kinetic
3.Ubuntu 16.04
4.64G RAM
5.8 core i7-9700 CPU @ 3.00GHz
6.GTX 1660

martins-mozeiko, 2019-09-26:
You could try reducing sensors you use on your vehicle. For example, do you need capture camera images? Check what sensors you use and try remove some of them, camera especially, as it also sends large amount of data.

Oasis-Guo, 2019-09-30:
After I switched to use ros_bridge from ROS repo instead of lgsvl/rosbridge_suit, the problem is solved

pedroexenberger, 2020-03-03:
@Oasis-Guo it's been a while since this issue were open but what do you mean by using ros_bridge from ROS? I'm facing a similar problem with version 2020.01, but I think it already uses rosbridge from ROS (/opt/ros/kinetic/.../rosbridge..). Was using a rosbridge from that path (from ROS) your solution? Or am I missing something?

Any help is appreciated. Thank you.

EricBoiseLGSVL, 2020-04-07:
@Oasis-Guo @pedroexenberger Yes, we are looking into updating our bridge implementation and any insight into solutions will help others and our team improve the code base.

pedroexenberger, 2020-04-07:
Hi @EricBoiseLGSVL, I created my own issue regarding this since I got no response at the time. I got a solution (for autoware.ai rosbridge and lgsvl) by adapting the ROS2 rosbridge code in LGSVL. You might check the post where I report that https://github.com/lgsvl/simulator/issues/653#issuecomment-595826566 as also the whole issue #653, which is closely related to this.

Best.

EricBoiseLGSVL, 2020-04-08:
@pedroexenberger Thank you for posting your findings.  This will help.  Sorry about the lack of response, we are currently cleaning up issues and adding tasks. This is a high priority right now