lgsvl/simulator #393: apollo 5.0   |-- default_end_way_point.txt  # Defined by FLAGS_end_way_point_filename not found!!

bestvcboy, 2019-09-27:
apollo 5.0   |-- default_end_way_point.txt  # Defined by FLAGS_end_way_point_filename not found!!

martins-mozeiko, 2019-09-27:
Could you clarify what do you mean by this? Does something not work? If it is error with Apollo, I suggest using their issue tracker to request support: https://github.com/ApolloAuto/apollo/issues

bestvcboy, 2019-09-27:
in lgsvl-apollo-5.0 
default_end_way_point.txt not found。

Map Data 

A bundle of correlated maps are organized as a directory, with a structure like:

<map_dir>                        # Defined by FLAGS_map_dir
  |-- base_map.xml               # Defined by FLAGS_base_map_filename
  |-- routing_map.bin            # Defined by FLAGS_routing_map_filename
  |-- sim_map.bin                # Defined by FLAGS_sim_map_filename
  |-- default_end_way_point.txt  # Defined by

bestvcboy, 2019-09-27:
in borregas_ave map！

martins-mozeiko, 2019-09-27:
This file contains pre-defined routing waypoints. You need to create depending on your use case.
When you run regular Apollo use-case (go to specified point on map) this file is not used.

bestvcboy, 2019-09-28:
I did the operation in APOLLO, but the path was not planned. I don't know how to start automatic navigation path planning.

martins-mozeiko, 2019-10-02:
Please following instructions on our documentation. It explains how to set route for Apollo to follow: https://www.lgsvlsimulator.com/docs/apollo5-0-instructions/#launching-apollo-alongide-the-simulator

closed 2020-04-07 by EricBoiseLGSVL