lgsvl/simulator #394: Error in Building from source file

Francisobiagwu, 2019-09-27:
I just cloned the source file using the git lfs install and git clone commands. I equally downloaded Unity 2019.1.10f for the simulator (In a nutshell, I just followed the instructions outlined in the building from source URL).
I am currently getting the following errors when I open the file in unity:
[image]
[image]

martins-mozeiko, 2019-09-27:
If you click `Clear` button, does these messages go away? If yes, then everything is fine. Sometimes Unity likes to show error messages which are not real. Usually this happens when opening project first time or after big update to code. I suspect they are showing errors while Editor is in the middle of opening project, but it has not fully loaded everything. So part of the code that has loaded complains that it cannot find a lot of stuff - but all that will be loaded later.

So just try clearing log, and running the project. If it works, then simple ignore these messages.

EricBoiseLGSVL, 2019-09-27:
It also helps to close the editor and re-open after the first time opening or after pulling a bunch of new changes.  @martins-mozeiko  is right, Unity can fail on start up sometimes.  Closing and re-opening will fix most issues.

Francisobiagwu, 2019-09-27:
Thanks for the response. I closed and reopened Unity and the error messages disappeared. I will like to camera and Lidar to the intersection. Where in the asset do I perform the edit? Also the map is missing from the asset

martins-mozeiko, 2019-09-27:
Lidar and camera code is in LidarSensor.cs and CameraSensor.cs files. But adding these components to something that is not car is not really support out of the box. It will require adding extra functionality to support this kind of feature. These sensors are created dynamically in AgentManager.cs. If you will want to create these components on their own, you'll need to check how it is done in AgentManager.cs and replicate its functionality to work without vehicle (mainly how OnBridgeSetup is called). Once they are created they will work automatically.

To have map inside editor you need to clone their repositories separately. We store map and vehicle assets separately from main repository to save size of git clone. You can read how to clone map and vehicle assets here: https://www.lgsvlsimulator.com/docs/assets/

Francisobiagwu, 2019-09-30:
Thanks Martins for the response. Is LG working towards adding this feature/functionality anytime soon, or should I go and figure it myself?

martins-mozeiko, 2019-10-02:
Currently we have no plans to implement such functionality.

closed 2020-04-07 by EricBoiseLGSVL