lgsvl/simulator #395: MapOrigin Documentation

moenasrallah, 2019-09-27:
What are the steps to get MapOrigin's parameters?
I have a custom map that I'm trying to match from LG with Apollo's HD Map.
Currently I just guess and run the simulator to get something close to matching with Apollo.
My guessed UTM's are not good for testing and want coordinates are close as possible... but I don't know which exact coordinates MapOrigin relies on

martins-mozeiko, 2019-09-27:
We typically do this in one of two ways:

1) if map is created from real data, like car driving around and collecting point cloud, camera images, gps data, then we know where in the real-world we started. This becomes (0,0,0) point in map recreated in Unity. So we take GPS coordinates of this point (from rosbag) and convert to UTM northing/easting values to put in MapOrigin.

2) more universal way is to look up map coordinates in some 3rd party map service, like Google Maps. You check what position corresponds to (0,0,0) point in your map inside Unity Editor (in top-down view). Then look up this point in Google Maps. Now you'll now latitude and longitude. These you convert to UTM northing/easting values and put in MapOrigin.

Make sure map you are creating is rotated correctly - map should be oriented so North is along the positive direction of global Z axis.

moenasrallah, 2019-09-27:
Okay thank you Martins. I imported the map via drone pictures and just stitched together a general OBJ, so #2 makes most sense in my case.

Closed.

closed 2019-09-27 by moenasrallah