lgsvl/simulator #396: potential bug in NPCController.SetChangeLaneData()

david-gwa, 2019-09-29:
hey, Lg-team,

        for lane change scenario,  with a different map,  where the wps indexes are not consistent in neighbor lanes, there is a problem  to  use `currentIndex` from previous lane to update  `currentTarget` for current lane (line891 in  NPController) :

 ````c#
    private void SetChangeLaneData(List<Vector3> data)
    {
        laneData = new List<Vector3>(data);
        currentTarget = laneData[currentIndex];  
        isDodge = false; // ???
    }

``` 

we are also looking for a more robost algorithms to set new  target in new lane

EricBoiseLGSVL, 2019-09-30:
@david-gwa 

Yes, you are correct.  We are adding some new features to NPCs and will look at a fix for this.  Thanks for letting us know.

DOMINICQQQ, 2022-06-30:
`#!/usr/bin/env python3
import lgsvl

# 创建模拟器
sim = lgsvl.Simulator(address = "localhost", port = 8181)

# 加载场景
sim.load(scene = "befe0755-ef70-4d81-aff4-354773f1c3d5", seed = None)

# 场景重置
sim.reset()

# 创建自车
ego_state = lgsvl.AgentState()
ego_state.transform.position = lgsvl.Vector(10, 0, 30)
ego_state.transform.rotation.y = 90
ego = sim.add_agent(name = "41095da1-3661-473e-a418-4d585652ea66", \
                    agent_type = lgsvl.AgentType.EGO, \
                    state = ego_state)
# 自车位置
s = ego.state
s.velocity.x = -50
ego.state = s
# 自车速度控制
c = lgsvl.VehicleControl()
c.throttle = 0.2
ego.apply_control(c, True)

# 运行模拟器
# 实时运行
sim.run(time_limit = 5.0)
# 非实时运行
# sim.run(time_limit = 6, time_scale = 2)`

DOMINICQQQ, 2022-06-30:
private void SetChangeLaneData(List<Vector3> data)
   {
       laneData = new List<Vector3>(data);
       currentTarget = laneData[currentIndex];  
       isDodge = false; // ???
   }

DOMINICQQQ, 2022-06-30:
使用个'git clone'

DOMINICQQQ, 2022-06-30:
use'git clone'

DOMINICQQQ, 2022-06-30:
use `git clone`

DOMINICQQQ, 2022-06-30:
```
   private void SetChangeLaneData(List<Vector3> data)
   {
       laneData = new List<Vector3>(data);
       currentTarget = laneData[currentIndex];  
       isDodge = false; // ???
   }

```