lgsvl/simulator #398: how to see main camera of vehicle in autoware rviz

clepz, 2019-09-29:
Hi all,
How can i see the main camera of vehicle in rviz? I have been trying to achieve this for a long time. I don't know what I did wrong. Can someone explain step by step how I can do that please?

BatuhanUlucay, 2019-09-30:
I can't display the vehicle camera in rviz, either.

Oasis-Guo, 2019-09-30:
1. U need to add a camera sensor as Json format from Web. See [color camera](https://www.lgsvlsimulator.com/docs/sensor-json-options/#color-camera)
2. I have try to modify the topic before, and it is not working at rviz. So I suggest U try to copy the json from example like Lexus2016RXHybrid (Autoware).
3. U can see the effect by toggling the visulization from simulator.
4. U should find the topic from rviz avaliable topic and choose the image. 
5. Should be working

BatuhanUlucay, 2019-09-30:
[image]

martins-mozeiko, 2019-09-30:
If I remember correctly rviz does not show compressed images. You need to use [image_transport](http://wiki.ros.org/image_transport) node in ROS to republish images as uncompressed RGB images in new topic, which you can then visualize in rviz.

clepz, 2019-10-01:
@martins-mozeiko thank you, I did it.
<b>Solution</b> is for people who have the same problem:
When lgsvl simulator connected to rosbrige, run this command `rosrun image_transport republish compressed in:=/simulator/camera_node/image raw out:=/new_topic_name`
It need to be /image , not /image/compressed careful!
Then don't forget to add topic in display -> add button -> by topic -> new_topic_name

closed 2019-10-01 by clepz

meitiever, 2020-03-16:
no image published still.

luqiang21, 2020-03-18:
@meitiever Did you see compressed image topics in `rostopic list`? If yes, have you followed @clepz 's solution?