lgsvl/simulator #399: No sensor data visulaised by dreamviewer

srirambadri, 2019-09-30:
Hi,

I followed the instructions [here](https://www.lgsvlsimulator.com/docs/apollo5-0-instructions/) to connect the simulator with apollo 5.0

I'm able to launch the simulator and connect it to apollo. When I manually move the vehicle in the simulation environment I see the model in the dreamviewer moving as well. So the connection seems to be working. 
But the problem is I don't see any of the information (sensor data) captured in the simulation environment in the dreamviewer screen. I've activated the necessary modules in the module controller tab but after few seconds they turn themselves off. 
[image]

Any idea as to what the problem could be?

martins-mozeiko, 2019-09-30:
Please check the logfiles of modules that turn off (Perception). It should contain more detailed error message.
Note that you do not need to enable Camera module in dreamview.

srirambadri, 2019-09-30:
I checked the apollo/data/log folder, there were no logfiles created by the perception module.Do you know where they're saved otherwise

martins-mozeiko, 2019-10-02:
They should be in this log folder. Check the permissions of folder, maybe it does not have permissions to write.

Alternative is to run perception in foreground, so all output shows up in terminal:
```
perception.sh start_fe
```

closed 2020-04-07 by EricBoiseLGSVL