lgsvl/simulator #401: ROSAgentManager doesn't exist.

clepz, 2019-10-01:
Hello everyone,<br>
I want to add a new vehicle, so I tried to follow the instructions on [this doc](https://www.lgsvlsimulator.com/docs/add-new-ego-vehicle/). However source code of the simulator doesn't have ROSAgentManager to add sensors in project. I've searched it <br>
What did I wrong?

hadiTab, 2019-10-01:
What version of the simulator are you using?

clepz, 2019-10-01:
v2019.09 <br>
I cloned the source code 4 or 5 days ago.

EricBoiseLGSVL, 2019-10-01:
@clepz 

We are in the process of updating our tutorials to the new HDRP simulator and they will release soon.

In the meantime, here are the steps.

1) Create a folder with the vehicle name in Assets/External/Vehicles/YourNewVehicle
2) Place create Models folder inside YourNewVehicle folder.  Import .fbx into Models folder, it should make a Materials folder inside Models folder.  If not, make one.  All materials and textures go in this folder.
3) Drag the Jaguar2015XE prefab that we provide into an empty scene.  This is for reference
4) Create a empty game object, name it YourNewVehicle and reset it's transform. Pos (0,0,0) Rot (0,0,0)
5) Drag the vehicle fbx as a child of this game object.
6) Right Click on the Jaguar2015XE object and choose Unpack Prefab.
7) Select the WheelColliderHolder, HeadLights, BrakeLights, IndicatorsLeft, IndicatorsRight, IndicatorsReverse, InteriorLight and Reflection Probe.
8) Drag these selected objects to YourNewVehicle so they are children of the root object.
9) Delete broken Jaguar2015XE prefab from scene and add a new one for reference.
10) Select root object of Jaguar2015XE.
11) Right click Copy Component in the Inspector panel on the RigidBody component
12) Select the YourNewVehicle root object and in the inspector panel, right click and Paste Component as New
13) Repeat for VehicleDynamics, VehicleController, and VehicleActions.
14) On the VehicleDynamics component, open Axles array.  Inside assign the FL FR RL RR wheel colliders and Left, Right, Front and Rear wheel meshes from your model.  See the Jaguar2015XE prefab for help.
15) Assign Tag "Player" and Layer "Agent" (All child objects)
16) Move lights and wheel colliders to match YourNewVehicle wheels and light positions.
17) Drag the root YourNewVehicle into the Project Panel in the Assets/External/Vehicles/YourNewVehicle folder.
18) At the top menu, Simulator -> Build... and make an asset bundle for that vehicle.  It places it in the repo AssetBundles folder.
19) Run the WebUI to then add a new vehicle and point it to the new vehicle. Boom, YourNewVehicle.

BatuhanUlucay, 2019-10-03:
Hi @EricBoiseLGSVL , I did all the steps you have mentioned, but when the simulation begins, the car does not stand still, it starts flying and falls down from the map as you can see below. I've added the logfile screenshot.

[image]
[image]
[image]

EricBoiseLGSVL, 2019-10-03:
@BatuhanUlucay 

I can tell immediately that the transforms are not reset.  Set the root prefab at position/rotation (0,0,0), scale (1,1,1) from step 4.  Left click the transform component top right gear and choose Reset.

[image]

Reset all child holder object as well.  Reposition the children of the child holders, e.g. the lights and wheel colliders.

[image]

Add a mesh collider to the body of the vehicle.

[image]

I have updated the 1-20 steps above to be more clear

BatuhanUlucay, 2019-10-04:
@EricBoiseLGSVL Thank you, adding the mesh collider to body did work.

closed 2020-04-07 by EricBoiseLGSVL