lgsvl/simulator #403: Cannot find NPCs in point clouds

huantc3033, 2019-10-02:
Version: 2019.07

Description: When adding LiDAR sensor to ego vehicle, if set visible we can see LiDAR points marked red in the environment, except the NPC vehicles. When subscribing to the topic of LiDAR sensor and visualize it with pcl visualizer, we cannot see any of them shaped like it is a vehicle.

Reproduction: Build the simulator and load the vehicles and maps as usual. Launch rosbrige_server and start running a simulation. Using a point cloud visualizer like rviz or others and subscribe to the topic of LiDAR. See the result displayed in rviz.

martins-mozeiko, 2019-10-03:
Can you please show screenshot of simulator where lidar points does not hit NPC vehicles?

huantc3033, 2019-10-29:
[image]
As the picture, there are no lidar points on the surface of NPCs, and as I open the pcl viewer, there seems no vehicles to be displayed.

martins-mozeiko, 2019-10-29:
This is very unusual. I'm not sure what is wrong here.

Which version are you using exactly? In first post you said 2019.07, but this version did not have sensor-view button (eye icon on the bottom) or the car selection on the bottom-right.

You mentioned that you have built simulator? Can you try running prebuilt binary form our official releases to see if it also happens?

closed 2020-04-08 by EricBoiseLGSVL