lgsvl/simulator #404: Error building WebUI

MithunKinarullathil, 2019-10-03:
Followed the [build instructions step by step](https://www.lgsvlsimulator.com/docs/build-instructions/). Getting this error below on step 23 on the latest sim.

ERROR: 

**Error building WebUI
UnityEngine.Debug:LogError(Object)
Simulator.Editor.BuildWebUI:Build() (at Assets/Scripts/Editor/BuildWebUI.cs:64)**

System:
OS: Ubuntu 16.04 LTS
CPU: Intel® Core™ i9-9900K CPU @ 3.60GHz × 16 
GPU: GeForce RTX 2080 Ti/PCIe/SSE2
Memory: 32 GB

MithunKinarullathil, 2019-10-03:
Partial solution: 

`sudo ln -s /usr/bin/nodejs /usr/bin/node`

Cause: [Issue](https://github.com/nodejs/node-v0.x-archive/issues/3911)

New Problem:

- You are running Node v4.2.6, but @enact/cli requires Node >=6.4.0.

Solution: [Node Js Upgrade](https://stackoverflow.com/questions/10075990/upgrading-node-js-to-latest-version)

- `sudo npm install n -g`

- `sudo n stable`

MithunKinarullathil, 2019-10-03:
Please mark this as solved. Build is complete now.

LOG: 

**WebUI build is completed
UnityEngine.Debug:Log(Object)
Simulator.Editor.BuildWebUI:Build() (at Assets/Scripts/Editor/BuildWebUI.cs:60)**

closed 2019-10-03 by MithunKinarullathil

twuser666, 2020-09-24:
> Followed the [build instructions step by step](https://www.lgsvlsimulator.com/docs/build-instructions/). Getting this error below on step 23 on the latest sim.
>
> ERROR:
>
> **Error building WebUI [...]

Hi, I have the same problem.
Can you teach me how to solve this problem?
Thank for your help.