lgsvl/simulator #406: Localhost:8080 shows empty (no MAp ,No vehicles,No simulations)

Iqbalparvi, 2019-10-03:
after a big fight as Build success but when i click on the Simulator it  bring me the localhost:8080 but there is nothing to run although-the npm install and  WebUI build also succes without any error:
[image] 
with this empty UI how we get simulator environment with map and vehicles 
is there any other possible mistakes that cause this issue? stuck here from two days kindly give  track 
thanks

martins-mozeiko, 2019-10-03:
If you are building simulator manually then you'll need to add environments and vehicles each individually. You can get links on our content website: https://content.lgsvlsimulator.com

Only when you run binary release first time, it will pre-create default vehicles, maps and some sample simulations. If you run it differently (manual build, or inside Unity Editor) will require doing this manually.

MithunKinarullathil, 2019-10-03:
@martins-mozeiko Would be a good idea to add this line of downloading sample scene and vehicle to the [build instructions](https://www.lgsvlsimulator.com/docs/build-instructions/)?

Iqbalparvi, 2019-10-04:
@martins-mozeiko Thanks I follow the Ist instruction by manually add enviroment and vehicle using the https:// url ........ its add enviroment (BorragasAVe)and vehicles(Lincoln MKZ 2017) then 
1.In simulation add new simulation it save as valid like:
[image]
we also see that there is no localhosht:9090 /bridge in the bridge section after selecting the map and vehicles:
[image]
2. when Click on the generated simulation  then it suddently become invalid as:
[image]
at the same time we see that the unity hub also just hang on simulator loading......
My target is this time is the only Binary release and required default vehicles ,maps and some simulations like (BorragAVe----linkolnn MKZ2017 Apollo-5.0) these are I found last time and suddently my system crash and i lose every things  ...so what is wrong with my simulation build , why i dont found these things default back ?
thanks

Iqbalparvi, 2019-10-06:
@martins-mozeiko I build the simulator using  assets manually  and see the vehicle (lincoln2017mkz) and map (BorragasAVE) but the simulator only shows the night sceen like: 
[image]
at the same time I run Apollo dramview to see the Above Simulator moving obstacle but My perception and traffic light trun off automatically so i cant see obstacles..
[image]
can you please guide me 
1. what is wrong with lgsvl why it shows night view enviroment ? 
2.why my perceptions and traffic light goes off and the car cant move automatically?
this problem stuck my whole research progress here and  waiting for suggestions 
to solve this 
regards

martins-mozeiko, 2019-10-07:
To change initial time of day, set it in simulation properties - see screenshot at bottom of this page: https://www.lgsvlsimulator.com/docs/simulations-tab/

To adjust time of day at runtime (for interactive simulation) do that in interactive menu that you access at bottom of window: https://www.lgsvlsimulator.com/docs/simulation-menu/#interactive-menu

To determine what is happening with perception & traffic-light modules check the log files under `apollo/data/log` folder. It creates separate file for each module. Most common reasons why these modules do not work is that GPU is not supported (Apollo currently supports only GTX 1080 or similar Pascal architecture models), or container is not run with nvidia docker. Just plain docker does not support forwarding GPU from host to container.

Iqbalparvi, 2019-10-07:
@martins-mozeiko thanks ,i go through the link to change the day time in both simulation and runtime  but problem cant solve : 
[image]
[image]
2ndly when i  run nvidia-smi before dev_into it shows all well like: 
[image]
but after dev_into there is no nvidia-smi found :  
and  there is no log files for these two modules(perception and traffic light) in data/log folders ....
so where i can do mistakes or left something.
regards

closed 2019-10-13 by Iqbalparvi

martins-mozeiko, 2019-10-14:
If you cannot run nvidia-smi, that means your nvidia-docker2 installation is not done correctly. Apollo uses `nvidia-docker` command to run its GPU container. If it does not find `nvidia-docker`, then it will fallback to regular `docker` command but that won't have access to GPU. Make sure you can run `nvidia-docker` command before running Apollo.

Iqbalparvi, 2019-10-14:
@martins-mozeiko thanks  ,aah i am confused here :
when i check nnidia-smi then it shows:
[image]
after installing nvidia docker and test using docker run --gpus all nvidia/cuda:9.0-base nvidia-smi
it shows : 
[image]
then when go inside docker as dev_into.sh then it shows nothing: 
[image]
so i guess the nvidia-docker is install  correctly, but the problem still remains same

martins-mozeiko, 2019-10-14:
You run `nvidia[space]smi`, but you should run `nvidia-smi` - delimited with dash.

Iqbalparvi, 2019-10-14:
ooh yes sory ,i run nvidia-smi but the it says nvidia-smi command not found 
[image]

martins-mozeiko, 2019-10-15:
That means nvidia-docker is not correctly installed.
Here you can see that Apollo checks if nvidia-docker command exists: https://github.com/lgsvl/apollo-5.0/blob/simulator/docker/scripts/dev_start.sh#L307
If it does not exist, it won't have GPU available inside container. Verify that `nvidia-docker` command exists on your host machine.

Iqbalparvi, 2019-10-15:
@martins-mozeiko  thanks ,,,,yes solve that problem to install nvidia docker when run nvidia-smi after dev_into it shows: 
[image]
closed this issue  now 
but related to the issue [https://github.com/lgsvl/simulator/issues/428] i still cant found any solution where i try to  bring the lgsvl daytime scene in simulation runtime