lgsvl/simulator #408: Unable to load any map Apollo 3.0 backend

andrewramacher, 2019-10-03:
When we start the backend Apollo server, we get these errors in the dreamview.INFO:

`I1003 13:45:26.234714 32198 map_service.cc:132] Failed to open map meta file: /metaInfo.json
E1003 13:45:26.234853 32198 file.h:98] Failed to open file /apollo/modules/map/data/simple_map/default_end_way_point.txt in text mode.
W1003 13:45:26.235333 32198 simulation_world_updater.cc:365] Failed to load default list of POI from /apollo/modules/map/data/simple_map/default_end_way_point.txt`

We get this error for both simple map and San Francisco. Inside the Docker, we cannot find the `default_end_way_point.txt` or `metaInfo.json` files. Is there somewhere we can get these?

hadiTab, 2019-10-03:
Can you explain the steps you are going through to get this error?

The `default_end_way_point.txt` file should not be needed for normal operation with the simulator.

andrewramacher, 2019-10-03:
We are trying to run the Apollo 3.0 backend with the steps listed here: https://github.com/lgsvl/apollo-3.0. The only difference is we have a different car (XE Rigged) and location (San Fransisco). When we go to the frontend, the main panel says 'Please send car initial position and map data' and never changes. Then when we check the dreamview.INFO file we see those errors. Any idea what could be causing this? Thanks!
[image]

hadiTab, 2019-10-04:
Just to confirm, you are using the old version of the simulator, right? Otherwise San Francisco and XE Rigged shouldn't be available.

Have you enabled the sensors in the simulator? You need lidar, gps and imu at minimum. If you want traffic light detection to work you also need cameras.

From the screenshot, it seems that either sensors are not enabled or rosbridge is not running.
Can you try opening a new terminal in docker (open new terminal and run the dev_into.sh script), and echoing the lidar data:
`rostopic echo /points_raw`

Let me know what the output is. Thanks.

andrewramacher, 2019-10-07:
Thanks for the response! Yes, we are running Apollo 3.0. Also, we had Localization, Perception, Planning, Prediction, Routing, and Control modules enabled, but the route editing tab was still unavailable and we could not move on from there. 

The output for `rostopic echo /points_raw` from the apollo directory within the docker was:

`/use_sim_time is not set, will not subscribe to simulated time [/clock] topic
WARNING: topic [/points_raw] does not appear to be published yet`

hadiTab, 2019-10-07:
This means that Apollo is not receiving lidar messages from the simulator.
Can you confirm that rosbridge is running by running `rosnode info /rosbridge_websocket` and pasting the output here?

Also just to confirm, are you enabling lidar in the simulator by clicking the check mark?

andrewramacher, 2019-10-08:
That is good to know, thank you. The output of `rosnode info /rosbridge_websocket` is as follows:
```Node [/rosbridge_websocket]
Publications: None

Subscriptions: None

Services: None

cannot contact [/rosbridge_websocket]: unknown node
```

And we have enabled lidar on the simulator, we tried checking all of the boxes.

hadiTab, 2019-10-08:
This means nothing is getting through to rosbridge. Does the status indicator in the top left of the simulator say 'Connected'?

andrewramacher, 2019-10-09:
Interesting, in the terminal where we are running ` ./scripts/rosbridge.sh` I can see the client connecting and disconnecting, and on the simulator it has 'Connected' in the top left corner.

EricBoiseLGSVL, 2020-04-08:
@andrewramacher We are no longer supporting old simulator, please update to latest for continued updates and fixes

closed 2020-04-08 by EricBoiseLGSVL