lgsvl/simulator #411: Unable to use LGSVL with Autoware 1.12

MaxandreOgeret, 2019-10-04:
Hello.

I am trying to use LGSVL with Autoware 1.12. I am following this piece of documentation : https://www.lgsvlsimulator.com/docs/autoware-instructions/#launching-autoware-alongside-lgsvl-simulator

But it appears that I am unable to load the launch files.
When loading the Map file I have many errors like this : 

``` 
[ERROR] [1570200341.001836157]: [createSignalMarkerArray] failed createPoleMarker: 4,4,6.75,0.4
```
But 'OK' is appearing next to 'ref' so I don know if itś critical or not.

Then when I load the sensing data I have this error : 

```
terminate called after throwing an instance of 'image_transport::TransportLoadException'
what():  Unable to load plugin for transport 'compressed', error string:
According to the loaded plugin descriptions the class image_transport/compressed_sub with base class type image_transport::SubscriberPlugin does not exist. Declared types are  image_transport/raw_sub
[republish-2] process has died [pid 1319, exit code -6, cmd /opt/ros/melodic/lib/image_transport/republish compressed in:=/simulator/camera_node/image raw out:=/image_raw __name:=republish __log:=/home/autoware/.ros/log/f6bef87a-e6b4-11e9-ba5c-9cb6d0c541d5/republish-2.log].
log file: /home/autoware/.ros/log/f6bef87a-e6b4-11e9-ba5c-9cb6d0c541d5/republish-2*.log
```

This error seems more serious and I don know how to solve it...

Do you have any idea ? Thanks a lot

hadiTab, 2019-10-04:
@MaxandreOgeret the first error is not critical and will happen every time. It's because we do not support all of the non-essential elements in the vector map. It will have no consequences.

The second error is related to the image transport node which is not available in the autoware container anymore. If you plan on using front facing camera this could cause issues. Otherwise, you can do basic operations without the front facing camera as long as lidar is working.

The easiest fix for this would be to add it to the container yourself.
The required package can be installed using apt:
`sudo apt install ros-kinetic-image-transport-pluggins`

You can either do this inside the container and then commit your changes using `docker commit`, or to edit the dockerfile and rebuilt the container.

MaxandreOgeret, 2019-10-07:
Thenks a lot for your answer ! I should have mentioned it but I am using **melodic** and not **kinetic**. And I can find any package named `ros-melodic-image-transport-pluggins` and I have these image packets installed on the docker image : 

 - `ros-melodic-image-geometry`
 - `ros-melodic-image-publisher`
 - `ros-melodic-image-transport`
 - `ros-melodic-image-view`
 - `ros-melodic-image-view2`

So I guess the plugins for `image-transport` are just not installed and not available on apt, even if it's status is `released` ...

According to [this](https://answers.ros.org/question/261378/image_transport-and-image_transport_plugins/) message I should compile it myself ?

MaxandreOgeret, 2019-10-07:
Thanks a lot for you help, I got it working ! :+1:

hadiTab, 2019-10-07:
I don't know what the reason is. It's possible that Autoware wasn't using it and decided to remove it.

codespjw, 2019-10-14:
@MaxandreOgeret  hello, I am trying to use LGSVL with Autoware 1.12. I am following this piece of documentation : https://www.lgsvlsimulator.com/docs/autoware-instructions/#launching-autoware-alongside-lgsvl-simulator, too.

However,  even after solving your reported issue, I cannot run the simulation successfluly.

I got the flowing errors

MaxandreOgeret, 2019-10-14:
I have the same error, but you should be able to follow the tutorial still.

closed 2020-04-08 by EricBoiseLGSVL