lgsvl/simulator #413: LGSVL simulator with ROS 2 bridge

weihunko, 2019-10-05:
### Issue summary
We are working on getting LGSVL simulator to work with ROS 2 web bridge, but we have encountered some issues. Here are the exact steps that we have tried:

1. Install ROS 2 dashing
   1. Follow these steps --> https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Install-Debians/
1. Install LGSVL simulator
   1. Download the latest release --> https://github.com/lgsvl/simulator/releases/tag/2019.09
   1. `sudo apt update && sudo apt install libvulkan1`
1. Install [ros2-web-bridge](https://github.com/RobotWebTools/ros2-web-bridge)
   1. Follow these steps --> https://github.com/RobotWebTools/ros2-web-bridge
   1. Install nodejs v10
      1. `curl -sL https://deb.nodesource.com/setup_10.x | sudo -E bash -`
      1. `sudo apt-get install -y nodejs`
   1. Run example --> https://github.com/RobotWebTools/ros2-web-bridge#run-examples
1. Start LGSVL simulator:
   1. Run `./simulator`
   1. Change the ROS bridge type of Lexus2016RXHybrid (Autoware) from ROS to ROS 2
   1. Run simulation "BorregasAve (with Autoware)"
1. In a new terminal
   1. `source /opt/ros/dashing/setup.bash`
   1. Run `ros2 topic list` --> can see the topic list (/points_raw, /odom, ...)
   1. Run `ros2 topic hz /points_raw` --> nothing shows up here
1. Try with rviz2 --> select the topic but nothing is visualized
1. We have also tried with ros1bridge and it worked. We can visualize the simulated pointcloud in rviz. 

### Screen shot of running simulator with ros2 web bridge
[image]

### Screen shot of inspecting using wirshark
[image]

Questions: 
1. Do you know which specific version of ros2-web-bridge work with LGSVL simulator?
1. Does any of the installation steps go wrong?

cyberphysicalrobotics, 2019-10-09:
1. We found the bug and fixed it. Latest version of ros2-web-bridge working with ROS2 Dashing will be working with LGSVL simulator. Please refer to email that I've sent. As for other users, next release of LGSVL simulator will work with ros2-web-bridge. 
2.The installation steps looks fine.

srirambadri, 2019-11-26:
Hi @cyberphysicalrobotics,

I'm facing a similar problem. 
Unfortunately I cannot install ROS2 Dashing on the system as the Ubuntu version installed is 16.04.

I have ros2-web-bridge installed  v0.2.7 and the latest patch of ROS2 Crystal.

[image]

Similar to @weihunko rostopic lists the topics correctly but I'm not able to see any data.

[image]
 
Do you know what the problem is?

cyberphysicalrobotics, 2019-11-27:
Hi, srirambadri When we tested it on Crystal, for some reason, ros2-web-bridge didn't work. We couldn't figure out what was the reason for that. I recommend testing it with ROS2 Dashing.

srirambadri, 2019-11-28:
@cyberphysicalrobotics do you by chance have a docker container with ROS Dashing.
I found one in the [lane following repo](https://github.com/lgsvl/lanefollowing/blob/master/Dockerfile) but it has crystal installed in it.

cyberphysicalrobotics, 2019-12-26:
$ echo $ROS_DISTRO 
crystal

$ ros2 topic list
/imu_raw
/lanefollowing/steering_cmd
/odom
/parameter_events
/points_raw
/rosout
/simulator/camera_node/image/compressed
/simulator/sensor/camera/center/compressed
/vehicle_cmd

$ ros2 topic echo /imu_raw
header:
  stamp:
    sec: 1577392389
    nanosec: 83340288
  frame_id: imu
orientation:
  x: -0.012414151802659
  y: -0.00639237649738789
  z: 0.992502868175507
  w: 0.121421135962009
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity:
  x: 1.09616667032242e-06
  y: -1.82492658495903e-06
  z: -1.1395239198464e-05
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration:
  x: 0.256974637508392
  y: -0.0948642194271088
  z: -9.80621814727783
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]

$ ros2 topic echo /points_raw

I tested lane following repo.
What topics I could see are:
/imu_raw
/odom
/simulator/camera_node/image/compressed

but there is topics which don't echo:
/points_raw
/simulator/sensor/camera/center/compressed

This question started because weihunko can't see /points_raw topic. ROS2 crystal have issue not to show some topics(/points_raw). So I recommend you to use ROS2 dashing instead of crystal.

closed 2020-10-06 by EricBoiseLGSVL