lgsvl/simulator #414: radar sensor supports autoware-car now?

nashdingsheng, 2019-10-06:
hi, i just want to confirm whether radar sensor supports autoware-car now?

martins-mozeiko, 2019-10-07:
Not in current source code. It would require adding code to publish correct message type for ROS that radar uses in Autoware. You can see [here](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Bridge/Cyber/CyberConversions.cs#L176) how it is done for CyberRT used by Apollo. You can add similar code for ROS Bridge and then everything would work.

nashdingsheng, 2019-10-07:
thank you for your reply, i have another question, except LGSVL,i have  been using prescan as a simulator  simultaneously. I feel prescan is convenient to use but it is of high price and  not open source.  LGSVL is under developing and i believe it will be more and more powerful as time elapses. I donot know whether it is challenging to modify the source code for my specific demand when I am not familar with c-sharp and unity programing skill. .
A specific demand is  whether the current version of LGSVL enbles to provides the vehicle status(yaw rate, steering angle, velocity,pos,orentation....) and many other informations  to autoware or apollo?   and how to ? or i need to modifythe source code myself.    my so-termed more information can be referenced by autoware_msgs. will LGSVL be made great efforts to develop continuously?

nashdingsheng, 2019-10-16:
I have a urgent question how can i use LGSVL if i do not have internet? i click the button to open the UI in the browser , but it fails  i think it fails because i do not have internet,thank you

zelenkovsky, 2019-10-16:
@nashdingsheng LGSVL Simulator depends on 3D content - environments and vehicles in order to run simulation. Content is distributed separately from simulator itself, this is why you need to create account and download content via internet at least once.

Make sure you've registered an account and keep cookies in your browser while you are offline. Otherwise it might ask you to login again.

BTW, you can make your own content as well, but it still requires you to have an account with LGSVL to run simulation.

closed 2020-04-08 by EricBoiseLGSVL