lgsvl/simulator #415: Occurred core dumped error  after run Bridge.sh

South-Jin-Shin, 2019-10-07:
Hi

I'm trying to run Apollo and link using bridge 
As a result, I succeeded running Apollo in browser (localhost:8888) 
but I failed to launch 'Bridge.sh' 

after command './bridge.sh' 
I can see the error below statements
=============================================================
WARNING: Logging before Init GoogleLogging() is written to STDERR
I1007 15:41:35.647729      363 global_data.cc:150] [cyber_bridge] host ip : [Own IP]
W1007 15:41:35.647786    363 Environment.h:33] [cyber_bridge] Environment variable [CYBER_PATH] not set, fallback to terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::system::system_error> >'
what() : bind: Address already in use 
./bridge.sh line 6:  363 Aborted 
(core dumped) bazel-bin/cyber/bridge/cyber_bridge 
=============================================================

I don't know how can i solve it... 

I already tried both root user and normal user

I think this error is probably a port configuration problem.
But I'm not sure how to set the port.

Wait for an answer...

martins-mozeiko, 2019-10-07:
`Address already in use` - this error means that somebody else on your machine is listening on port 9090. Find out what program uses it and close it. Alternative is to modify bridge.sh script to use different port.

You can see which program is using port 9090 by executing following command:
```
sudo ss -ltnap | grep 9090
```

South-Jin-Shin, 2019-10-07:
@martins-mozeiko 

I checked that use 'ss' command and 'netstat' command and then found the process using 9090 port that status is LISTEN 

clearly killed processes 

and then I tried executing  ./bridge.sh 

As a result, I got 3 lines like below captured output and then nothing work without any response (so I stopped process using ctrl+z) but other error lines disappeared 
> WARNING: Logging before InitGoogleLogging() is written to STDERR
> I1007 22:23:22.011714 9885 global_data.cc:150] [cyber_bridge] host ip: [$ Own IP]
> W1007 22:23:22.0111780 9885 environment.h:33] [cyber_bridge]  Environment variable [CYBER_PATH] not set, fallback to 

P.S : I don't know how to change 9090 port, please let me know how to change a port for bridge

closed 2020-04-08 by EricBoiseLGSVL