lgsvl/simulator #416: Building & Compiling Error

srirambadri, 2019-10-07:
Hi,

I'm having an issue when I try to build Apollo 5.0.
[image]

I'm using Ubuntu 16.04 and I've tried to build the original fork of the apollo platform and it builds successfully.

martins-mozeiko, 2019-10-07:
How are you compiling Apollo? Inside docker or outside?

srirambadri, 2019-10-07:
Inside
I followed the instruction: https://www.lgsvlsimulator.com/docs/apollo5-0-instructions/

srirambadri, 2019-10-07:
I looked at the apollo.sh file and this is where the build fails ("at 2. build failed")
[image]

srirambadri, 2019-10-07:
I managed to build 100% but I commented some lines in the apollo.sh file
[image]

the problem is that I'm not able to see and sensor data on the dream viewer 
[image]

lemketron, 2020-08-16:
Is this still an issue? Might try apollo master as the lgsvl changes have been migrated upstream, though the lgsvl fork of apollo 5.0 seems to be building and working fine for most folks lately...

closed 2020-09-25 by lemketron