lgsvl/simulator #418: How to fix error *No module named 'lgsvl', after running python API for running LG simulator?

Rizwan-Abdullah, 2019-10-07:
I am trying to run python API(example) by following the procedure stated in running the LGSVL simulator for the autonomous driving scenario but it is giving me an error saying no module found lgsvl.

I am running the code as well as the lg simulator in the python folder. python version 3.6 is used.

hadiTab, 2019-10-07:
Did you install the python api using pip?
You need to run `pip3 install --user -e .` in the PythonApi folder.

Another possible issue is that you have more than one python3 installed on your system and pip is installing it for a different python3 than the one you intend to use it for. To test this you can try opening a python interpreter for the other python3 on your system and seeing if you can `import lgsvl` there. If so, I'd suggest using a virtual environment.

Rizwan-Abdullah, 2019-10-08:
thankx for replying
i only have 1 python version so that option is out. i have tried the first option by installing through pip in PythonAPI folder but still it is not working. i have attached the screen shot kindly look to see to find any problems
[image]

Rizwan-Abdullah, 2019-10-08:
the problem is solved. i had to install in PythonAPI and the dot at the end of the command is very important for the command to run. so it is pip3 install --user -e.

closed 2019-10-08 by martins-mozeiko