lgsvl/simulator #420: RViz 2D Nav goal with LGSVL confusing and not working

MaxandreOgeret, 2019-10-08:
Hello everyone.

I am trying to follow this tutorial : https://www.lgsvlsimulator.com/docs/autoware-instructions/

I have set up everything and the link is working perfectly, I can make the car follow a set of pre-recorded waypoints perfectly.

But I can manage to follow a path from the vector map using the `2D Nav Goal` functionnality of RViz and the tutorial is lacking some important details. (I'll be very happy to make a PR to add more details to it).

[image]

My main issue is this part under 'Driving by following vector map:' : 

 > Enable `waypoint_loader` while making sure the correct route file is selected in the app settings

The tutorial should give more info about the "correct route file" because I don't know how to find it. 

 - Is RViz supposed to generate it when I use `2D Nav Goal` ?
 - Is it a CSV file ?
 - Is it supposed to to be in the `/tmp/` directory (default directory for the `waypoint_loader`) ? Because I can find any `.CSV` file in this directory.

[image]

I really need help with this. Thanks a lot for everything !

MaxandreOgeret, 2019-10-08:
Problem is solved and you need to load the `lane.csv` file.

hadiTab, 2019-10-08:
@MaxandreOgeret that is a mistake in the documentation. We have fixed it in our local repo and we will be pushing it to github soon.

You actually do not need to choose any csv file. That step was copied over from somewhere else. You can just omit that step completely.

MaxandreOgeret, 2019-10-08:
Thanks a lot for your help, but then if the waypoint_loader is not used what is loading the waypoints generated in RViz ?

Thanks !

hadiTab, 2019-10-08:
The HDmap internally has waypoints in it which the planner is aware of and used for planning the route. I believer the planner also publishes the points you see in rviz.

If you use waypoint_loader it will also show some waypoints in rviz along with velocity at each waypoint. In general, these are just visualizations are not important for the operation of the autonomous stack.

closed 2020-04-08 by EricBoiseLGSVL

tekrajchhetri, 2020-07-06:
@hadiTab have you pushed the changes? I am facing the same issue https://github.com/lgsvl/simulator/issues/842. If not how to fix it?