lgsvl/simulator #422: Can't run any PythonAPI quickstart examples that use raycast.

mmarco9771, 2019-10-08:
I get the following error whenever i run any of the examples which include sim.raycast():

  File "C:\Users\Beni\Anaconda3\lib\site-packages\lgsvl-0.0.0-py3.7.egg\lgsvl\simulator.py", line 216, in raycast
    if hit[0] is None:

TypeError: 'NoneType' object is not subscriptable

-------------------------
LGSVL version: 2019.09.
PythonApi version: lgsvl.0.0.0

I'm running the simulator from the windows executable (not building from Unity). My pyApi version looks weird, but other examples not using raycast work correctly.

hadiTab, 2019-10-08:
@mmarco9771 It seems like you have cloned the latest python API from master. If that is the case, master includes changes that were not part of the 2019.09 release (which your binary build is from). Please download and use the version of the pythonAPI for the release (you can find them [here](https://github.com/lgsvl/PythonAPI/releases)).

Yes, we should fix the versioning of the PythonAPI in the setup.py file as well. Thanks for the catch.

mmarco9771, 2019-10-08:
Thanks for the quick response, cloning from master was indeed the source of the problem.

closed 2019-10-08 by mmarco9771