lgsvl/simulator #423: installation issue with Unity Hub

hyunjaecho94, 2019-10-09:
I have the same problem as thread 293 (https://github.com/lgsvl/simulator/issues/293), which is that Unity Hub fails to add the Simulator folder in its Projects tab (step 9 in the installation page). Just thought I would create a new issue because thread 293 has been dormant for a while. Tried the recommendations there, like logging out/in from Unity Hub, creating a new empty project,  removing UnityHub folder at ~/.config/UnityHub, but still cannot resolve the problem. When I try to open the project directly with Unity, it automatically opens Unity Hub, so I could not proceed with the installation.

EricBoiseLGSVL, 2019-10-09:
@hyunjaecho94 
Have you updated Unity Hub?  This issue was fixed by updating.  What version are you using / OS?

Iqbalparvi, 2019-10-10:
i also facing issue in step 12 in the installation link https://www.lgsvlsimulator.com/docs/build-instructions/
here when i click to INSTALL--->LOCATE--->no UNity folder found here as u can see below:
[image]
whats wrong with my installation to unity 2019.1.10f1, why the UNITY---> unity 2019.1.10f1 folder not generate here
thanks

martins-mozeiko, 2019-10-10:
@Iqbalparvi Unity folder is here:

[image]

Iqbalparvi, 2019-10-10:
@martins-mozeiko thanks ,thats right so the instruction 12 select the launcher from unity folder means is this ....
[image]
if it is the right option then why the simulator cannot change the day time scene (always remains on night mode although i set daytime 12:00,2:30 3:00 etc ) and why the simulator runtime window  automaically trun off(close/disable). 
 
[image]

hyunjaecho94, 2019-10-10:
upgrading Unity Hub solved the problem. Thank you.

closed 2019-10-10 by hyunjaecho94