lgsvl/simulator #424: About ray tracing support

yumianhuli2, 2019-10-10:
Unity has released a version of ray tracing,Has ray tracing been supported?Thank U!

martins-mozeiko, 2019-10-10:
If I'm not mistaken, the DXR support in Unity is still in preview, not in stable version. So nothing has changed from situation mentioned in #364.

That said, if you are interested in working on this, we would gladly accept pull request.

Closing issue, as it is duplicate of #364. Please continue discussion in one place.

closed 2019-10-10 by martins-mozeiko

yumianhuli2, 2019-10-11:
> If I'm not mistaken, the DXR support in Unity is still in preview, not in stable version. So nothing has changed from situation mentioned in #364.
>
> That said, if you are interested in working on [...]

ok thanks!