lgsvl/simulator #428: LGSVL simulator runtime disappear suddently

Iqbalparvi, 2019-10-13:
Hi LG;
After build succes the standalone it show two differnt error for me:
1. the simulator run but it shows the dark enviroment only although i canhge gthe build time to 12:00 ,1:30 etc and there is no play button  found in the simulator:
[image]
2. when connect the bridge and run apollo -5.0 fork then open the modules transform the running simulator suddently disappear 
i s there any problem in the unity loader/launcher to load the scene ,,, need sugestions to solve this problem asap
thanks

martins-mozeiko, 2019-10-14:
1.

Play button is visible only to interactive simulation. For non-interactive time will always be running forwards.

Are you saying environment is always black regardless of what time you set? 01:00, 12:00, 18:00, 23:00?

2.

This could be related to running out of GPU memory issue. Check the log file. If at the end it has similar call stack to log file in #326 then your machine does not have enough memory to run BorregasAve together with Apollo. Read my reply on that issue.

Iqbalparvi, 2019-10-15:
thanks @martins-mozeiko  
[image]
yes i change the time in localhost :8080 during the simulation setting but there is no option to change the time in runtime simulation and no play button : 
[image]
and for GPU memory when we run nvidia-smi we see that volatile  GPU util is 0% so it means it is out of memory 
[image]

martins-mozeiko, 2019-10-15:
Play button and time changing functionality is enabled only for **interactive** simulation. Put checkbox next to interactive setting: https://www.lgsvlsimulator.com/docs/images/full_size_images/web-simulation-mapVehicle.png

Iqbalparvi, 2019-10-15:
@martins-mozeiko  ,the play button and time changing functionality found but the darkness remains same and not change by changing the time .....
2ndly the runtime simulator suddently disappear ....? when i  check the gpu consemption using 
sudo watch nvidia-smi  then its shows the GPU consumtion increase slowly from 0 to 100
[image]

 and as it reach 100 % the simulator disappear ...how to fixed this issue without losing the previous works ..
thanks

martins-mozeiko, 2019-10-15:
Because your GPU has only 6GB of memory I'm pretty sure this is because of running out of memory.

We recommend using GPU with 8GB or more if you want to load high quality maps such as BorregasAve or AutonomouStuff. These maps uses high detail textures that uses a lot of memory. Especially when you want to run simulator together with Apollo on same machine.

Iqbalparvi, 2019-10-15:
@martins-mozeiko Shocking ...I recently purchased this system  for my experiment purpose and at the initial stage only apollo run very good in this system but when we combine  LG then at the ist time it work good but bit slow  at low graphic mode ....as we make some progress it suddenly shows this termination problem and my RAM is 16 GB as per my understanding NVIDIA GFORCE GTX 1060 : 
[image]
but when the command lspci -v -s 00:02.0 it gives 
[image]
is there any proper way to confirm the GPU information ...really shocking at this stage

martins-mozeiko, 2019-10-15:
16GB is your regular memory amount, not GPU memory amount. nvidia-smi shows amount of GPU memory you have: 

[image]

Simulator itself should be fine with 6GB of memory, but if you run something else, like Apollo or Web browser with a lot of tabs, then that uses more memory, so simulator will run out of memory. Apollo itself uses significant amount of memory for its GPU perception (DNN networks).

closed 2020-04-08 by EricBoiseLGSVL