lgsvl/simulator #434: Attaching blue lines for autonomous Apollo use

ninjacoder0, 2019-10-16:
Hi,
I am trying to do a lane merge scenario. How do I attach the to blue lines/What is the best approach so that in apollo I could autonomously go beyond what is shown in image 2. and get the vehicle to successfully merge, thanks  

[image]

[image]

luqiang21, 2019-10-16:
@ninjacoder0 I think the car doesn't go beyond the point in image 2 is due to the current lane is not connected with the succeeding lane. I suggest you to validate this by viewing the `txt` file of your map file. To convert a `.bin` to `.txt` to view, using following command inside apollo docker 

`protoc --decode=apollo.hdmap.Map modules/map/proto/map.proto --proto_path=. < path_to_your_base_map.bin > base_map.txt`

Display the lane ID in dreamview by clicking the corresponding button in `Layer Menu`, then look for it inside the `base_map.txt`, check whether it has `successor_id` or not. 

Make sure the lanes which need to be connected their end points are overlapping with each other. Let me know whether my assumption is true.

daohu527, 2019-10-17:
Additional, I think the problem is the left lane is at the middle of the right lane. The lgsvl calc lane algorithm will not deal with this problem. So I suggest: 
Make the right lane into two lanes. the end of right lane is the same of the left lane. Then you draw a new lane, which start with the end of the right lanes. Be sure the end point and the start point less than 1meter. Then check the lane which @luqiang21 suggest.

luqiang21, 2019-10-17:
@daohu527 You are right! I didn't notice this. @ninjacoder0 You need to break the long lane into two parts @daohu527 said, then you should have connections correctly.

shalinmehtalgsvl, 2019-10-18:
To expand on this, @ninjacoder0 you will need to break up the annotations into 2 parts:

Part 1 (the merge):
1 MapLaneSection with the 4 lanes, 3 lanes in the highway and the incoming lane. The end of this section should have the last waypoint of the merging lane and the last waypoint of the left lane overlapping.

Part 2 (after the merge):
1 MapLaneSection with the 3 lanes of the highway. Continues until an intersection or the number of lanes changes.

closed 2020-04-08 by EricBoiseLGSVL