lgsvl/simulator #435: planar Lidar

alexsimon1, 2019-10-16:
hello, 

if I modify the RayCount of the Lidar to 1, the Lidar dont work anymore.
I use the LGSVL Version 2019.04

Thank you for your help.

hadiTab, 2019-10-16:
@alexsimon1 can you provide more details on what is not working?
Is it not changing the number of beams? Is it not publishing to ROS?
I just ran a quick test with Apollo and it seemed to work fine with a single beam.

alexsimon1, 2019-10-16:
I cant see the beam in the simulator and i dont get any data from the ros topic. I used the autoware model. Maybe it’s because i used the old version?

hadiTab, 2019-10-16:
Is it publishing at all? i.e. what is the output of `rostopic echo /points_raw`?

alexsimon1, 2019-10-18:
There is no output at all on the topic. Only If i change the laser count higher then 1 the data is published.

hadiTab, 2019-10-18:
@alexsimon1 I'm not sure why that is happening, but I don't have that issue with the current version of the simulator. If you want to use the old simulator, we do have a [planar lidar sensor](https://github.com/lgsvl/simulator/blob/2019.04/Assets/Scripts/LidarSensor/PlanarLidarSensor.cs) in the old simulator which was used for the tugbot robot. The lidar publishes a [LaserScan](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/LaserScan.html) message though (not pointcloud). You can ass this lidar to your ego vehicle and use it instead of the multi-beam lidar.

alexsimon1, 2019-10-18:
Ok thanks for your answer, I will try the tugbot.

closed 2019-10-18 by alexsimon1