lgsvl/simulator #436: Build Maps&Vehicles

yumianhuli2, 2019-10-17:
Hello!I downloaded CubeTown  SingleLaneRoad and Jaguar2015XE  and put them into External/Environments and Vehicles folders , then I clicked Build button there is  no reaction from unity.How about this?
[image]

martins-mozeiko, 2019-10-17:
Is there nothing printed out in console? Error or warning message?

Try providing path where to build simulator. Currently you have this field completely empty.

daohu527, 2019-10-18:
You should choose a folder to save the build files. This option is located in the "Build Simulator" of the image.

closed 2020-04-08 by EricBoiseLGSVL