lgsvl/simulator #437: Delay between Measurement Timestamp and Message Timestamp

christian-lanius, 2019-10-17:
Hi everyone,
Thanks for your work on the simulator. I am trying to use it with Apollo 5.0 and I have a problem with the localization module: There is a timestamp issue between the measurement and the message timestamp:

[image]

We can increase this delay arbitrarily, simply by pausing the simulation using the pause button in the interactive mode. After continuing the simulation, the delay has increased, not by the total pause time, but approx. proportional to it.

hadiTab, 2019-10-18:
@christian-lanius I'm currently looking into this and hopefully should have a fix soon.
To give some context, the timestamp 1255344916.973999977 is in the GPS time system and refers to October 17, 2019. It seems like there is a mix up in the time for measurement and timestamp for GPS related topics.

ntutangyun, 2020-02-01:
[image]

this is a live screenshot I just took.

Thanks to @hadiTab 's explaination, now I understand that the `/apollo/sensor/gnss/best_pose` is in GPS time system.

however, I'm using Apollo 5.0 fork, and it's clear that `/apollo/localization/pose` message (published by Localization module) is always several seconds ahead of even the module's input `gnss/imu` etc. 

I wonder if anyone is encountering same issues ? Thank you.

ntutangyun, 2020-02-03:
I changed how simulator manager computes current time and managed to reduce above mentioned lag from 7 seconds to about 0.05 seconds. 

below is the update
```c#
    void FixedUpdate()
    {
        // CurrentTime += Time.fixedDeltaTime;
        CurrentTime = (DateTime.UtcNow - unixEpoch).TotalSeconds;
        CurrentFrame += 1;

        if (!IsAPI)
        {
            PhysicsUpdate();
        }
    }
```
I'm not sure what other problems this would bring. Thus haven't submitted a pull request yet.

Please help verify this.  Thanks

closed 2020-10-20 by EricBoiseLGSVL