lgsvl/simulator #440: Missing Simulator.exe

hyunjaecho94, 2019-10-18:
In step 2 of testing simulator after installation, it says to double-click Simulator.exe. However, I cannot find the file. Which step in my installation could have gone wrong?

martins-mozeiko, 2019-10-18:
What installation steps are you referencing?
Simulator.exe is inside release zip file that you can download from [Release](https://github.com/lgsvl/simulator/releases) tab in GitHub.

hyunjaecho94, 2019-10-19:
I was referencing https://github.com/lgsvl/simulator/blob/master/Docs/docs/build-instructions.md. Step 2 under Test Simulator (Double-click the Simulator.exe that was built).

hyunjaecho94, 2019-10-19:
So I downloaded the zip file and was able to get Simulator.exe. However, I'm not sure what step 3 of the installation page (under Test Simulator) means by "Select graphics options then press Ok". Here's what I get from Simulator.exe.

[image]

martins-mozeiko, 2019-10-21:
simulator.exe file is from zip file for Windows. You need to download Linux zip file if you want to run Simulator on Ubuntu.

closed 2019-10-28 by hyunjaecho94