lgsvl/simulator #442: Simulations Failed

yumianhuli2, 2019-10-20:
Hello!
Map is ok!
[image]
Vehicle is ok!
[image]
But can not select bridge!
[image]
Finally,Simulation mode test invaild!
[image]
And I built AssetBundle successed!
[image]

How can I run simulation sucessful?thank you!

martins-mozeiko, 2019-10-20:
Please check the [log file](https://www.lgsvlsimulator.com/docs/faq/#where-are-unity-log-files-located) for more detailed error message.

yumianhuli2, 2019-10-21:
> Please check the [log file](https://www.lgsvlsimulator.com/docs/faq/#where-are-unity-log-files-located) for more detailed error message.

[image]

martins-mozeiko, 2019-10-21:
You should start simulation from Web interface and when it fails, look at bottom of file for any error messages. Without out more detailed error message it is very hard to guess what could be wrong.

yumianhuli2, 2019-10-23:
> You should start simulation from Web interface and when it fails, look at bottom of file for any error messages. Without out more detailed error message it is very hard to guess what could be wrong.
just say this..without more detailed error message 
error message just :+1: System.ArgumentException: Kernel 'KSampleCopy4_1_x_8' not found.

[image]

closed 2020-04-08 by EricBoiseLGSVL