lgsvl/simulator #443: build-instructions issue

ningfeiwang, 2019-10-21:
Hello,

I am trying to build the lgsvl simulator in Linux following the build-instructions(https://github.com/lgsvl/simulator/blob/master/Docs/docs/build-instructions.md).  However, when I do step 7(In the Projects tab, click Add and select the folder that the Simulator was cloned to),  nothing was added in the Unity Hub without any hints. I try several times, even re-launch step 1, but still cannot add anything.

My system is Ubuntu 16.04. Unity Hub is 2.1.2.

Could you please help me to handle such issues? 

Really appreciate your time. Thank you very much.

yumianhuli2, 2019-10-21:
> Hello,
>
> I am trying to build the lgsvl simulator in Linux following the build-instructions(https://github.com/lgsvl/simulator/blob/master/Docs/docs/build-instructions.md). However, when I do step [...]

You can create a empty project in unity ,and then select the simulator project.

martins-mozeiko, 2019-10-21:
Yeah, try creating new project and then adding existing project should work.
Seems to be very popular Hub bug: https://forum.unity.com/threads/cant-add-existing-project-with-unity-hub-2-0-0-if-there-arent-currently-any-projects.688354/

ningfeiwang, 2019-10-21:
Hello,

Thanks for your reply. I also tried it. However, after creating an empty project in Unity Hub, Unity Hub, itself, will turn off automatically. When I check the new project in the corresponding path, I found nothing.

Is that caused by my OS or Software itself? Not sure whether this issue could be handled.

ningfeiwang, 2019-10-28:
Thank you. It works after I update my system and Unity Hub.

closed 2019-10-28 by ningfeiwang