lgsvl/simulator #444: Minimum PC requirements for LGSVL/ Apollo 5.0 Fork

ioanniskyr, 2019-10-21:
We are interested in coming up with a distributed Apollo 5.0/LGSVL setup where Apollo 5.0 runs on PC1 and LGSVL on PC2, but we are not quite sure about the minimum hardware spec requirements each PC should have and also about the list of incompatible HW. For example, we have read that the current version of Apollo does not support Volta and Turing architectures. Our intention is to run this setup purely in simulation.

Could you please provide a list of minimum requirements for the Apollo and the LGSCV PCs?

Any help would be greatly appreciated.

ioanniskyr, 2019-10-21:
Thank you very much for your reply. So, I guess Apollo supports GeForce GTK, but not RTK? Does it have to be GTX 1080 specifically? For example, what about GTX 1660?

martins-mozeiko, 2019-10-21:
1660 will not work as it is same architecture as 20xx GPU's - Turing.

ioanniskyr, 2019-10-25:
Thank you very much, I really appreciate your help.

closed 2019-10-25 by ioanniskyr

Martin20, 2020-04-29:
Hello,

I am going to buy a machine with RTX 2080 GPU. Since you mentioned in the link below that it is not suitable for apollo 5.0 simulation, do you have any update showing now apollo 5.0 works with that GPU?

https://github.com/lgsvl/apollo-5.0

**The current version of Apollo does not support Volta and Turing architectures (this includes Titan V and RTX 2080 GPUs).**

lemketron, 2020-04-30:
At this point you would want a machine with a Nvidia GTX 1070 or 1080 GPU for Apollo 5.0. That might be a desktop or laptop depending on your needs.

Note that Apollo 5.5 will run on the newer (Touring architecture) RTX 2070 and 2080 GPU but Apollo 5.5 does not yet work with the LGSVL Simulator. We are looking into Apollo 5.5 support, but cannot say at this time when that will be available.

zqzqz, 2020-05-13:
Hi @lemketron I have a confusion here. Apollo provided an option of CPU build. Is it possible to run Apollo with CPU and connect it to lgsvl? Thanks!

lemketron, 2020-05-13:
> Hi @lemketron I have a confusion here. Apollo provided an option of CPU build. Is it possible to run Apollo with CPU and connect it to lgsvl? Thanks!

You can run Apollo on CPU but enabling perception requires a GPU. With Apollo 5.0 that must be a GTX-1070, 1080, or similar (Pascal), but with Apollo 5.5 and newer you can also use the RTX 2070, 2080, or similar (Touring).

See issue #614 for more information on our (in-progress) support for Apollo 5.5.