lgsvl/simulator #445: screen blinking

hyunjaecho94, 2019-10-22:
When playing the simulator, I experience blinking of the entire simulator. I am using Ubuntu 16.04, GeForce GTX 1080. I tried lowering the graphics quality, but that does not seem to be the issue. Has anyone else experienced this and solved it?

video zip 
[blinking-2019-10-22_12.55.50.mkv.tar.gz](https://github.com/lgsvl/simulator/files/3757261/blinking-2019-10-22_12.55.50.mkv.tar.gz)

EricBoiseLGSVL, 2019-10-22:
@hyunjaecho94 

We are not able to see the video you linked, can you link again or make a gif?  Are you running the Editor or build?  If you are running the Editor, there is a known Unity bug if you have the Game panel open and the Scene panel open at the same time.  If you dock them both so only one is active at a time, it will prevent the flicker.

If you are using a build, please post more details so we can help.

hyunjaecho94, 2019-10-22:
I'm not sure what you mean by graphics driver. The result of `sudo lshw -c video` is the following.

[image]

I am using the latest version of the simulator.

shalinmehtalgsvl, 2019-10-22:
Can you share the output of `nvidia-smi` as well? That has the version number of the driver.

christian-lanius, 2019-10-23:
I am facing the same problem as well. I can fix it by setting the camera and rendering resolution to a very low value (i.e. 640x480). The same problem exists for both a newly compiled version and for the precompiled binary. @hyunjaecho94 can you check the log if there is an error "Failed to read GPU texture" coming from the lidar sensor? These two problems coincide for me I think.
I am running 
Kubuntu 18.04
RTX 2080 TI
nvidia driver 418.67
```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 418.67       Driver Version: 418.67       CUDA Version: 10.1     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce RTX 208...  On   | 00000000:01:00.0  On |                  N/A |
| 35%   35C    P8    16W / 260W |    544MiB / 10986MiB |      2%      Default |
+-------------------------------+----------------------+----------------------+
```

EricBoiseLGSVL, 2019-10-23:
@hyunjaecho94 @christian-lanius 

We have seen the Failed to read GPU texture and have been looking into the cause.  We will continue to debug.  Thanks for letting us know about this.

pranavbajoria93, 2020-02-28:
I am facing the same problem. I am using the rosbridge fork and getting an alternating connected-disconnected error in the rosbrisge terminal and a flickering screen in the simulator. 

I am not using docker images but have the simulator and ROS melodic running in an Ubuntu 18.04 system. I am using a GM200 [GeForce GTX TITAN X]. My drivers seem to be up to date. Here is my Nividia SMI

[image]

In my error log, I get a Failed to read GPU texture just as @christian-lanius mentioned. Do you need any more information.
Is there any work around to this? 

> I can fix it by setting the camera and rendering resolution to a very low value (i.e. 640x480).
@christian-lanius I get an option to change my screen resolution when I start my simulator. Where do I change the camera and rendering resolution in the unity project before hitting the play button?

I am exploring the simulator and still learning, any help is appreciated.

EricBoiseLGSVL, 2020-02-28:
@hyunjaecho94 @christian-lanius @pranavbajoria93 We are still looking into this issue.  This could be a driver issue.  I would try re-installing the driver.  This issue may help [287](https://github.com/lgsvl/simulator/issues/287)

pranavbajoria93, 2020-03-04:
@EricBoiseLGSVL I have tried reinstalling drivers and made sure that the vulkan drivers are correctly installed. But I am still unable to get a stable connection over ROS bridge. I have mentioned the issue in detail in https://github.com/lgsvl/simulator/issues/658. Could you please see what I might be missing.

EricBoiseLGSVL, 2020-03-04:
We are looking into updating Unity to see if it helps with the resolution issue.  @pranavbajoria93 I have pinged someone on the team that may be able to help with ROS issues

EricBoiseLGSVL, 2020-04-08:
@hyunjaecho94 @christian-lanius @pranavbajoria93 We are releasing an update to unity that solves many of these Vulkan issues.

closed 2020-04-08 by EricBoiseLGSVL