lgsvl/simulator #447: Build ERROR with Unity Editor

clouddd9, 2019-10-24:
I download the simulator folder and open it by unityHub,.However, the Unity Editor didn't load the simulator correctly and cast errors as follow:
[image]
[image]

why did it happen & what should I do to fix the problem?

martins-mozeiko, 2019-10-24:
That looks like git lfs is not installed or is installed correctly. Please verify that git-lfs is installed correctly.
https://www.lgsvlsimulator.com/docs/faq/#why-are-assets-scenes-missing-empty-after-cloning-from-git

clouddd9, 2019-10-25:
I'm pretty sure that I've installed the lfs in the cmd and it shows "Git LFS initialized."
However, the problem still exists.
I found that when I input bridge.sh in docker, it shows (which can be seen in the picture I posted before)

WARNING:logging before InitGoogleLogging() is written to STDERR

and in Unity3D the errors all show in this form:
Assets/Scripts/ApolloAutoMode.cs(2,7): error CS0246: The type or namespace name `xxxxx' could not be found. Are you missing an assembly reference?

Is it the WARNING that  leads to the problem?

martins-mozeiko, 2019-10-25:
Bridge has nothing to do with this. Bridge is relevant only when simulator runs with vehicle. Currently simulator cannot even start, because code can not compile due to errors.

Can you please check `Assets/Plugins/Protobuf/protobuf-net.dll` file on your disk? It should be ~269KB in size. If it is only 100 or so bytes, then this is git lfs issue. Try running `git lfs ls-files` - all files should have asterix (*) next to them. If it does not have asterix, then git lfs is not working out your checkout. Try fetch and checking out files again (`git lfs fetch` and `git lfs checkout`).

clouddd9, 2019-10-25:
well, the size is about 131KB.
In what path should I run the command? I running in /home/user but the feedback is 

cloud9@cloud9:~$ git lfs ls-files
Not in a git repository.

clouddd9, 2019-10-25:
Oh, I noticed that I didn‘t git the simulator. I just download the .zip 
Very sorry to bother you. Now I'll try the git version and see what's changed.

martins-mozeiko, 2019-10-25:
131KB sounds strange. Did you mean just B as in Bytes, not KB (kilobytes)? Then this is git-lfs issue.

GitHub does not package files under git-lfs storage inside zip file downloads from repository: https://github.com/isaacs/github/issues/712
If you want to work on simulator source code, you will need to do `git clone ...`

clouddd9, 2019-10-25:
my Unity edition is 2019.1.10f1
and I think I've installed the node.js correctly cuz when i input node it shows
<code >
cloud9@cloud9:~$ node  
</code>
<code >
node      node-gyp  nodejs  
</code>
and git lfs have been installed too.
I follow the guide in everything I did and here come the errors.

martins-mozeiko, 2019-10-25:
These are not real errors. Press `Clear` button and they will go away.
Unity likes to complain with errors about some of its internal workings (maybe due to fact that Linux platform is not very well supported). But these errors are harmless. Everything else still works. nodejs has nothing to do with this.

clouddd9, 2019-10-29:
I finally realized that it's empty in <code>Assets/External/Environments</code>and <code>Assets/External/Vehicles</code>.  And that's why I can't choose the environments and vehicle models when building. THX.

closed 2019-10-29 by clouddd9