lgsvl/simulator #448: Driving from point A to B by giving waypoints in dreamview

hyunjaecho94, 2019-10-25:
Inside the simulator, when I manually move the vehicle, the vehicle in the simulator also moves, suggesting the simulator is connected to dreamview. However, when enabling Localization, Transform, Traffic Line, Planning, Prediction, Routing, and Control, Perception and Traffic Line  automatically get switched off. When I give waypoints under Route Editing, the vehicle in the simulator does not move. No NPC vehicles or sensor data are visible. Here are some information that may be helpful. Related: @399
`git lfs install` > `Git LFS initialized.`
`nvidia-smi`
[image]

`./docker/scripts/dev_into.sh`
[image]

`./apollo.sh build_gpu`
[image]

Result of ` perception.sh start_fe`

`derek@in_dev_docker:/apollo$ bridge.sh
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1025 15:38:17.647433 21515 global_data.cc:150] [cyber_bridge] host ip: 172.27.44.58
Could not create logging file: No such file or directory
COULD NOT CREATE A LOGGINGFILE 20191025-153817.21515!Assets/Plugins/Protobuf/protobuf-net.dll
Could not create logging file: No such file or directory
COULD NOT CREATE A LOGGINGFILE 20191025-155643.21515!Could not create logging file: No such file or directory
COULD NOT CREATE A LOGGINGFILE 20191025-155714.21515!Could not create logging file: No such file or directory
COULD NOT CREATE A LOGGINGFILE 20191025-155741.21515!Could not create logging file: No such file or directory
COULD NOT CREATE A LOGGINGFILE 20191025-155741.21515!^[[A^Cderek@in_dev_docker:/apollo$ perception.sh start_fe
mkdir: cannot create directory '/apollo/data/log': Permission denied
[cyber_launch_25005] INFO Stop cyber launch finished.
[cyber_launch_25005] INFO cyber_launch exit.
[cyber_launch_25041] INFO Stop cyber launch finished.
[cyber_launch_25041] INFO cyber_launch exit.
[cyber_launch_25077] INFO Stop cyber launch finished.
[cyber_launch_25077] INFO cyber_launch exit.
derek@in_dev_docker:/apollo$ [cyber_launch_25441] INFO Launch file [/apollo/modules/drivers/tools/image_decompress/launch/image_decompress.launch]
[cyber_launch_25441] INFO ========================================================================================================================
[cyber_launch_25441] INFO Load module [image_decompress] library: [mainboard_default_25441] [CYBER_DEFAULT] conf: [/apollo/modules/drivers/tools/image_decompress/dag/image_decompress.dag] exception_handler: []
[cyber_launch_25441] INFO Start process [mainboard_default_25441] successfully. pid: 25496
[cyber_launch_25441] INFO ------------------------------------------------------------------------------------------------------------------------
[cyber_launch_25442] INFO Launch file [/apollo/modules/perception/production/launch/perception_all.launch]
[cyber_launch_25442] INFO ========================================================================================================================
[cyber_launch_25442] INFO Load module [perception] library: [perception] [CYBER_DEFAULT] conf: [/apollo/modules/perception/production/dag/dag_streaming_perception.dag] exception_handler: []
[cyber_launch_25440] INFO Launch file [/apollo/modules/transform/launch/static_transform.launch]
[cyber_launch_25440] INFO ========================================================================================================================
[cyber_launch_25440] INFO Load module [static_transform] library: [mainboard_default_25440] [CYBER_DEFAULT] conf: [/apollo/modules/transform/dag/static_transform.dag] exception_handler: []
[cyber_launch_25440] INFO Start process [mainboard_default_25440] successfully. pid: 25504
[cyber_launch_25440] INFO ------------------------------------------------------------------------------------------------------------------------
[cyber_launch_25442] INFO Start process [perception] successfully. pid: 25503
[cyber_launch_25442] INFO ------------------------------------------------------------------------------------------------------------------------
[cyber_launch_25442] INFO Load module [perception_camera] library: [perception] [CYBER_DEFAULT] conf: [/apollo/modules/perception/production/dag/dag_streaming_perception_camera.dag] exception_handler: []
[cyber_launch_25442] INFO Load module [perception_traffic_light] library: [perception_trafficlights] [CYBER_DEFAULT] conf: [/apollo/modules/perception/production/dag/dag_streaming_perception_trafficlights.dag] exception_handler: []
[cyber_launch_25442] INFO Start process [perception_trafficlights] successfully. pid: 25507
[cyber_launch_25442] INFO ------------------------------------------------------------------------------------------------------------------------
[cyber_launch_25442] INFO Load module [motion_service] library: [motion_service] [CYBER_DEFAULT] conf: [/apollo/modules/perception/production/dag/dag_motion_service.dag] exception_handler: []
[cyber_launch_25442] INFO Start process [motion_service] successfully. pid: 25513
[cyber_launch_25442] INFO ------------------------------------------------------------------------------------------------------------------------
[mainboard_default_25441]  WARNING: Logging before InitGoogleLogging() is written to STDERR
[mainboard_default_25441]  I1025 16:04:05.703177 25496 module_argument.cc:81] [] command: mainboard -d /apollo/modules/drivers/tools/image_decompress/dag/image_decompress.dag -p mainboard_default_25441 -s CYBER_DEFAULT 
[mainboard_default_25441]  I1025 16:04:05.703603 25496 global_data.cc:150] [] host ip: 172.27.44.58
[mainboard_default_25441]  I1025 16:04:05.704185 25496 module_argument.cc:57] [] binary_name_ is mainboard, process_group_ is mainboard_default_25441, has 1 dag conf
[mainboard_default_25441]  I1025 16:04:05.705906 25496 module_argument.cc:60] [] dag_conf: /apollo/modules/drivers/tools/image_decompress/dag/image_decompress.dag
[perception]  WARNING: Logging before InitGoogleLogging() is written to STDERR
[mainboard_default_25441]  Could not create logging file: No such file or directory
[perception]  I1025 16:04:05.718248 25503 module_argument.cc:81] [] command: mainboard -d /apollo/modules/perception/production/dag/dag_streaming_perception.dag /apollo/modules/perception/production/dag/dag_streaming_perception_camera.dag -p perception -s CYBER_DEFAULT 
[perception]  I1025 16:04:05.724697 25503 global_data.cc:150] [] host ip: 172.27.44.58
[perception]  I1025 16:04:05.726498 25503 module_argument.cc:57] [] binary_name_ is mainboard, process_group_ is perception, has 2 dag conf
[perception]  I1025 16:04:05.727236 25503 module_argument.cc:60] [] dag_conf: /apollo/modules/perception/production/dag/dag_streaming_perception.dag
[perception]  I1025 16:04:05.727530 25503 module_argument.cc:60] [] dag_conf: /apollo/modules/perception/production/dag/dag_streaming_perception_camera.dag
[perception]  Could not create logging file: No such file or directory
[perception]  COULD NOT CREATE A LOGGINGFILE 20191025-160405.25503!E1025 16:04:05.781314 25503 class_loader_utility.cc:220] [mainboard] poco LibraryLoadException: libcuda.so.1: cannot open shared object file: No such file or directory
[perception]  E1025 16:04:05.781997 25503 class_loader_utility.cc:236] [mainboard] poco shared library failed: /apollo/bazel-bin/modules/perception/onboard/component/libperception_component_lidar.so
[perception]  E1025 16:04:05.782310 25503 class_loader_manager.h:70] [mainboard] Invalid class name: SegmentationComponent
[perception]  E1025 16:04:05.782534 25503 module_controller.cc:59] [mainboard] Failed to load module: /apollo/modules/perception/production/dag/dag_streaming_perception.dag
[perception]  E1025 16:04:05.782826 25503 class_loader_utility.cc:258] [mainboard] Attempt to UnloadLibrary lib, but can't find lib: /apollo/bazel-bin/modules/perception/onboard/component/libperception_component_lidar.so
[... 40 lines cut ...]
[perception_trafficlights]  
[cyber_launch_25442] ERROR Process [perception] has finished. [pid 25503, cmd mainboard -d /apollo/modules/perception/production/dag/dag_streaming_perception.dag /apollo/modules/perception/production/dag/dag_streaming_perception_camera.dag -p perception -s CYBER_DEFAULT].
[cyber_launch_25442] ERROR Process [perception_trafficlights] has finished. [pid 25507, cmd mainboard -d /apollo/modules/perception/production/dag/dag_streaming_perception_trafficlights.dag -p perception_trafficlights -s CYBER_DEFAULT].
[... cut ...]
[mainboard_default_25440]  E1025 16:04:07.070777 25797 node_manager.cc:101] [mainboard] this process will be terminated due to duplicated node[static_transform], please ensure that each node has a unique name.
[mainboard_default_25440]  
[cyber_launch_25440] ERROR Process [mainboard_default_25440] has finished. [pid 25504, cmd mainboard -d /apollo/modules/transform/dag/static_transform.dag -p mainboard_default_25440 -s CYBER_DEFAULT].
[cyber_launch_25440] INFO All processes has died.
[cyber_launch_25440] INFO Cyber exit.
[cyber_launch_25440] INFO All processes have been stopped.
`

martins-mozeiko, 2019-10-25:
Permission denied errors probably comes from a fact that you have run dev_start.sh with sudo previously. This means Apollo created folders with root user. And now your user cannot write to these folders or files. Try running `chown` on all apollo files & folders to reset owner to your user/group.

Is first screenshot of nvidia-smi from inside of docker container or on host? Verify that nvidia-smi works also inside container.

hyunjaecho94, 2019-10-26:
@martins-mozeiko 
`nvidia-smi` does not work inside the container..

martins-mozeiko, 2019-10-28:
Please check output of `docker info | grep Runtimes` command on your host. It should show:

    Runtimes: nvidia runc

Also you should have `nvidia-docker` executable available (Apollo will try to use it). If you don't have these two conditions, then something is not installed correctly with nvidia docker. If I remember correctly you need to install nvidia-docker2 package.

Make sure docker correctly works with nvidia runtime before continuing with building Apollo & using Simulator.

hyunjaecho94, 2019-10-28:
Progress: `docker info | grep Runtimes` worked, `nvidia-docker` was giving command not found. Followed instructions here https://github.com/nvidia/nvidia-docker/wiki/Installation-(version-2.0) to get `nvidia-docker` working and perception is now working on Dreamview. Dreamview detects other NPC vehicles. 

However, when given a waypoint on Dreamview, a route is generated, but EGO does not move. Also, before perception worked, when EGO was in the way of other NPC vehicles, they would crash into EGO, but now NPC stops if EGO is in the way. Does enabling perception also allow perception for NPC vehicles?

martins-mozeiko, 2019-10-28:
NPC crashing into EGO or not is just a random behavior. Apollo perception does not affect NPC vehicles.

We sometimes see that Apollo vehicle does not start driving automatically. If it has path set and Control module is enabled, try helping it a bit - press Arrow up key to manually move car for short amount of time. After that Apollo should automatically take control.

hyunjaecho94, 2019-10-28:
@martins-mozeiko I tried that, but EGO just moves straight. 

[dreamview.tar.gz](https://github.com/lgsvl/simulator/files/3780921/dreamview.tar.gz)

hyunjaecho94, 2019-10-28:
works perfectly now. Very much appreciate it!

closed 2019-10-28 by hyunjaecho94