lgsvl/simulator #449: Feature discuss: need a routing command in python api?

daohu527, 2019-10-28:
In the process of running the simulator with apollo using python api, we first run python api, then switch to dreamview and use "route editing" to set the routing destination. it's too much trouble.

I hope to send the routing command directly from python api, so that I only need to open apollo, then use python api to run the whole process. The simulator can be more automated.   
I have implemented the features, is this a good idea?

martins-mozeiko, 2019-10-28:
We probably don't want any AD specific API in our Python API that is meant for communicating with simulator. It probably should be separate package and it should abstract AD stack so it can be used with other AD stacks. If you have any specific requirements or design for this kind of API, it would probably be useful to post it, so anybody can discuss.

We ourselves are using small Python code to interact with Dreamview over websocket to perform basic actions. Not sure if that is the best way. There probably is a better way to integrate with Apollo for controlling everything more precisely.

daohu527, 2019-10-29:
I recognize your point of view, python api try to do general functions. My design is to add a navigation sensor in ego car. The python api sets the navigation by send command to the navigation sensor.
The whole process: python api -> simulator(navigation sensor) -> apollo(or other AD). I don't know if autoware is like this, but I think if it is an api for integration test, if you need to control 2 softwares, it is really too much trouble to use. 
The navigation module is an general function.

closed 2020-04-08 by EricBoiseLGSVL