lgsvl/simulator #450: Textures pink, not loaded

KrzysztofDabek, 2019-10-28:
Hi,

I have just downloaded and run the simulator in unity editor with this tutorial:
https://www.lgsvlsimulator.com/docs/build-instructions/

After testing simulation all the textures are not loaded (pink or black) until I manually change shader type and change it back.

Is there a way to re-compile all the shaders or something similar? Is it a problem with HDRP?

martins-mozeiko, 2019-10-28:
Are textures pink on vehicle or whole map?
How are you loading vehicle and map? From asset bundles you built or downloaded from our website? Note that these bundles are OS specific. Loading Linux bundle on Windows (and vice versa) will result in missing shaders - thus the pink color. Make sure you are building & using asset bundles for correct OS.
In next version we'll have platform independent bundles, so there will be no differences between Windows and Linux.

KrzysztofDabek, 2019-10-29:
I have re-downloaded the asset bundles and made sure that the files I'm using are Linux specific.
Now the shaders and materials have been loaded but they do not contain textures. Basically everything looks default like there is no colour or texture or map on it.