lgsvl/simulator #455: Can I control NPC behavior on Dreamview?

hyunjaecho94, 2019-10-29:
I want to test some edge cases. As of now, NPCs are spawned and move randomly (respecting traffic rules). Is there a way to control them such that I can generate some scenarios?

martins-mozeiko, 2019-10-29:
You can move NPCs by making them follow waypoints. You can pass exact coordinates to `follow` api call.
See [13-npc-follow-waypoints.py](https://github.com/lgsvl/PythonAPI/blob/master/quickstart/13-npc-follow-waypoints.py) example script.

hyunjaecho94, 2019-10-29:
yes, but as far as I know, that is available using Python API. However, I cannot run Apollo's AD stack using Python API without using ROS perhaps. EGO only moves by keyboard input or giving it starting velocity/steering at the generation part of the scene. 

Whereas in Dreamview, Apollo 5.0 runs on the EGO vehicle but I don't seem to have control over NPC vehicles. I am asking if there is a way to control NPCs in Dreamview.

martins-mozeiko, 2019-10-29:
You can create EGO vehicle with Python API and connect it to Apollo.
22-connecting-bridge.py shows how to connect it to bridge.
Then Apollo will be able to control the vehicle.

Alternative to Python API is to write custom C# code inside Unity to control NPCs.

hyunjaecho94, 2019-10-29:
What do you mean Apollo will be able to control the vehicle? Could you please provide an example of how Apollo controls the vehicle?
Also, when running `python 22-connecting-bridge.py`, the connection does not seem to be established:

[image]

rosbridge.sh is running inside docker in the background.

martins-mozeiko, 2019-10-29:
You need to specify correct IP address inside script for it to know where is bridge running. If it is same machine as python script, then put it 127.0.0.1 or localhost.

Apollo will control EGO vehicle same way how it does when you run simulator without Python API. As long as bridge is connected - the sensors will send messages and you'll see localization working inside Dreamview. Then you'll be able to use Dreamview to set route to use.

hyunjaecho94, 2019-10-29:
I am using the same machine as the python file. I do not change the default IP address of 127.0.0.1 and leave the port number to default value of 8181. However, I still receive an error:
[image]. The ports for Dreamview and the simulator are 8888 and 8080 respectively.

In the meantime, rosbridge outputs 
[image]
which hopefully gets resolved after the connection is established.

Finally, to clarify, in the last paragraph of your response, do you mean that I can give waypoints to the EGO vehicle through Dreamview, and inside Python API, it can receive and follow those waypoints while respecting Apollo's AD stack?

I really appreciate you helping me out.

martins-mozeiko, 2019-10-29:
I meant that you need to change IP address in argument of `connect_bridge` call. It needs to have IP address where bridge is running. It needs to have same value you are usually putting in WebUI interface when you are creating simulation instance. [This line](https://github.com/lgsvl/PythonAPI/blob/master/quickstart/22-connecting-bridge.py#L29)

Basically you can run this setup on three machines:
A - where you run Python script
B - where simulator runs
C - where AD stack + rosbridge runs

You need to pass IP address of machine B in argument to `lgsvl.Simulator` And you need to pass IP address of machine C in argument to `connect_bridge` call.

hyunjaecho94, 2019-10-30:
Thank you. I was able to run 22-connecting-bridge.py by changing line 29 to 127.0.0.1.

> Simulator won't receive these waypoints. Only AD stack will know waypoints & route to follow. AD stack will send back control commends (accelerate, brake, steering value) to simulator which will move the vehicle.

What would the process to move the EGO vehicle this way be like? I am thinking that is should be along the line of the following,
1. Run rosbridge in background
2. Select API Only mode in simulator
3. Run a Python API and connect to bridge
4. Open Dreamview, select Vehicle, Map
5. Under Module Controller, enable Localization, Transform, Perception, Traffic Light, Planning, Prediction, Routing, and Control
6. Under Route Editing, add and submit waypoints
7. Apollo will receive waypoints and route to follow and send control commends (accelerate, brake, steering value) to simulator which will move the vehicle.

But since the API ONLY mode of the simulator is open, Route Editing in Dreamview is grayed out.

martins-mozeiko, 2019-10-30:
That sound correct. But for route editing to work, Apollo should be able to localize vehicle & receive ROS/Cyber messages. This means you need to start simulation time. So somewere after point 3 and before 6 you need to call `sim.run()` in Python script. After that you'll be able to continue with point 6.

hyunjaecho94, 2019-10-31:
Excellent!! It works perfectly. Final question: at stop signs, the vehicle stops, and I would have to nudge it a little with keyboard to keep the vehicle moving. Is this a normal behavior, or is there a way to fix this?

martins-mozeiko, 2019-10-31:
If map annotation is correct then vehicle should automatically continue after stopping before stop sign.

hyunjaecho94, 2019-10-31:
I'm sorry. It was not a stop sign but a stop due to red traffic light. Even if the light turns green, the vehicle does not move until I give it a little nudge forward. I am using the provided Borresgas Ave map.

martins-mozeiko, 2019-10-31:
For traffic lights Apollo will move forward only when it can detect that traffic light is green. If it does not see traffic light - maybe perception or traffic light module is not running, then it won't move.

hyunjaecho94, 2019-10-31:
Right. Traffic light toggles off itself even though perception works fine. Where could that have gone wrong?

martins-mozeiko, 2019-10-31:
Check the log file for traffic light output. It should have error message about why it is turning off.

hyunjaecho94, 2019-10-31:
In `/apollo-5.0/data/log`, there is no traffic light log, maybe because it is not working? Below are what I have: 
conti_radar.INFO        
mainboard.INFO
control.INFO            
map.INFO
cyber_bridge.INFO       
monitor.INFO
dreamview.INFO          
perception.INFO
gnss.INFO               
planning.INFO
image_decompress.INFO 
 prediction.INFO
kv_db_tool.INFO         
routing.INFO
localization.INFO      
 static_transform.INFO
mainboard_default.INFO

martins-mozeiko, 2019-10-31:
Not sure which file it logs to, but you can run Traffic Light module manually in command line with following command:

    mainboard -d modules/perception/production/dag/dag_streaming_perception_trafficlights.dag

This will print all log messages to terminal.

hyunjaecho94, 2019-10-31:
I am guessing it should be run inside the docker. I disconnected out of rosbridge and ran the command. Here's what I get:

[image]

martins-mozeiko, 2019-10-31:
Yes, it should be run inside Docker. mainboard is Apollo tool they made to run different modules.
Segmentation fault means that process crashed. Most often this happens because of nvidia issues. Either driver is not correctly installed, or nvidia-docker does not work. You'll need to load coredump into gdb to analyze call stack to understand what exactly happened. Sometimes simple reboot of computer helps. But this is really Apollo issue. Simulator or bridge has nothing to do with it. You can also run bridge in background by appending `&` symbol to command.

I suggest to search similar issues on Apollo issue track, maybe they have more details for this kind of error: https://github.com/ApolloAuto/apollo/issues

hyunjaecho94, 2019-10-31:
I will try that. Thank you so much for all your support.

closed 2019-10-31 by hyunjaecho94