lgsvl/simulator #457: LGSVL build from source issue

ningfeiwang, 2019-10-30:
Hello,

I am working on build the newest version LGSVL from source following the instruction https://www.lgsvlsimulator.com/docs/build-instructions/

When I go into step 23 (Build WebUi... in Unity: Simulator -> Build WebUI...), I found out 
[image]

Could you please help me to handle such problems? Thank you

My system is Ubuntu 18.04.

martins-mozeiko, 2019-10-30:
This is harmless error message. It comes from some internal components of Unity and we cannot do much about it. But it does not prevent anything to work. Simulator will run just fine. Press `Clear` button in corner to get rid of this error message.

ningfeiwang, 2019-10-30:
Thank you. I just clear it and continue building it after clone the maps and vehicles. But I met this problem. Could you please help me again?

[image]

martins-mozeiko, 2019-10-30:
You need to install Windows support when you are installing Unity. When Simulator builds assets we build "universal bundle" - one file that works both on Windows and on Linux. For this to work you need to have installed both - Windows and Linux support when installing Unity.

KrzysztofDabek, 2019-10-30:
I have coped with that kSampleCopy problem just as in this topic on unity:
https://forum.unity.com/threads/kernel-ksamplecopy4_1_x_8-not-found-under-unity-2018-3.597463/

Reimporting HDRP worked. Also please make sure you have vulkan libs installed as in the building from source tutorial

closed 2019-11-09 by ningfeiwang