lgsvl/simulator #459: How to create/modify vehicles

KrzysztofDabek, 2019-10-30:
Hi,

I have seen that there is a tool for creating new maps.
Is there any tool to create or modify vehicles as asset bundles?

martins-mozeiko, 2019-10-30:
Currently we do not have official instruction how vehicle should be created.
But you can refer to instructions in comment from Eric here: https://github.com/lgsvl/simulator/issues/401#issuecomment-537265124
Comments in issue #441 also could help.

KrzysztofDabek, 2019-10-31:
I have followed the instructions in Eric's comment.
Unfortunately the Build creates an assetbundle in format that is no longer valid:
the zip file with linux, windows and textures. Or at least in my case.

martins-mozeiko, 2019-10-31:
You need to update simulator code or download latest binary. Latest release introduced new format for maps and vehicles, which is, as you already said, zip file.

KrzysztofDabek, 2019-11-05:
I have updated my code. The building option in Unity still creates old format Asset bundle.
Also. How do I edit existing map as an asset bundle?

martins-mozeiko, 2019-11-05:
Could you please verify that [Build.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Editor/Build.cs#L339) file on your local disk looks like this around line 339?

    using (ZipFile archive = ZipFile.Create(Path.Combine(folder, $"environment_{manifest.assetName}")))

KrzysztofDabek, 2019-11-06:
There are errors about missing data to create textures.
Which files are mandatory to create an asset bundle?

KrzysztofDabek, 2019-11-06:
I have managed to create assetbundles from models found in your github but if I change anything in the map (SingleLaneRoad), everything turns black as if there was no light in the scene.
Also, when I build a vehicle (Jaguar), I get multiple errors:
[image]
but the vehicle is built

martins-mozeiko, 2019-11-06:
By everything turns black - do you mean this happens at runtime, or inside editor where you are changing map? Because inside editor there is no lighting in scene, so everything is expected to be black. We add light (sun) dynamically when scene is entering play mode. Workaround for this inside editor is to turn off lighting in Scene view - lightbulb icon at top of Scene window.

KrzysztofDabek, 2019-11-07:
[image]
At runtime after uploading AssetBundle. The only thing I have changed was the rotation of the map.
At the same time SingleLaneRoad from contents website works fine, even with custom vehicles

martins-mozeiko, 2019-11-07:
Can you check what time of day is set for scene? Inside Unity Editor you can check SimulatorManager -> EnvironmentEffectsManager and check "Current Time Of Day" property. Try changing it up and down to see if anything changes.

For player, you need to run scene in interactive mode, then you'll have same slider available in menu item at bottom of screen: https://www.lgsvlsimulator.com/docs/images/full_size_images/interactive-menu.png

enisotomata, 2019-11-08:
Thank you very much, it worked in simulation but made no change in unity viewport.

KrzysztofDabek, 2019-11-08:
I had CurrentTimeOfDay set to 12. When I set it to 0 it was a little bit brighter but still dark

enisotomata, 2019-11-08:
if you do it in unity, it is not changing.You must do it in the simulator and a value of 20 worked for me.

KrzysztofDabek, 2019-11-08:
@enisotomata You are right. Thank you for that. That is only a workaround but it's ok.

The bigger question is why does it behave like that? It seems as if day and night have been switched, because 18 works like 6 should and so on.
Seems like 0:00 is still midnight but 23:30 is almost noon.

zelenkovsky, 2019-11-09:
Latest simulator calculates precise sun position based on GPS location of your map and time-zone.
Check MapOrigin and timezone.

KrzysztofDabek, 2019-11-12:
What do you mean by that? Does 12:00 not mean 12:00 if we build the map in a different zone than Western America?

martins-mozeiko, 2019-11-13:
Time for map should be in timezone of map. Not in your local timezone. Timezone for map is set in MapOrigin component inside the scene. So regardless of where you are the sun position should be same inside the map.

Few weeks ago there were some fixes for this code, so make sure you are using up to date release or source code. If it still does not work, then that is bug.

I believe there still was a bug for setting time of day when you run scene from editor directly, not going through WebUI loader. That should be fixed in our next weekly push.

EricBoiseLGSVL, 2020-04-07:
@KrzysztofDabek We are finishing a step by step tutorial for creating new environments and will be releasing it soon

closed 2020-04-07 by EricBoiseLGSVL