lgsvl/simulator #461: Simulator isn't able to start, although I choosed 'API only' in Localhost

RamoramaInteractive, 2019-11-02:
After downloading the LGSVL Simulator Repository I startet it and in the Localhost I've choosed 'API Only' and clicked on 'run'. The windows showed me the message 'API ready', but nothing happens.

I want to start the simulator, but nothing works. The simulations says 'Invalid: You must specify a valid AssetBundle'. There is nothing I can do to download the assets. The download will always be interrupted.

All maps says 'Invalid: Missing or wrong Map AssetBundle. Please check content website for updated bundle or rebuild the bundle.'.

I'm using a computer with Windows 10 on a Intel Core i7-6700, NVIDIA GeForce GTX960 2GB and DDR4 16GB.

martins-mozeiko, 2019-11-03:
"API ready" means that simulator is waiting for Python API calls. When you run "API" simulation it means you want to run Python script that will control all aspects of simulation. Which includes - map loading, EGO vehicle loading, NPC placement, simulation time, etc... I suggest you read our documentation on Python API for more details: https://www.lgsvlsimulator.com/docs/python-api/

Other error messages probably means that you have run Simulator previously and now you have old version of maps and vehicles. Latest version of simulator change map & vehicle file format. So you need to download newer ones. Copy & paste Asset Bundle link from map & vehicle you want to use from our [content](https://content.lgsvlsimulator.com/) website and put them over existing links in Web interface.

RamoramaInteractive, 2019-11-03:
Thank you, that was very helpful! :)

About "API ready": I didn't know, that it was for the Python API. I thought it was only another kind of simulation.

closed 2020-04-08 by EricBoiseLGSVL